gabonator/LA104 #4: Cannot build dfutool on ubuntu

endeavour, 2019-04-02:
I tried to run build.sh on ubuntu 18.10 and I get the following error:

```
./build.sh 
In file included from dfuload.cpp:13:
fat12.h:190:10: fatal error: i386/endian.h: No such file or directory
 #include "i386/endian.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
In file included from fat12tool.cpp:13:
fat12.h:190:10: fatal error: i386/endian.h: No such file or directory
 #include "i386/endian.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
```

gabonator, 2019-04-02:
Just comment out that include line. And please let me know if it helped

endeavour, 2019-04-02:
Thanks for quick reply. Now I get a lot more errors:
```

dfuload.cpp: In function ‘int root_dir_addr(int, bpb33*)’:
dfuload.cpp:20:5: error: ‘uint32_t’ was not declared in this scope
     uint32_t offset;
     ^~~~~~~~
dfuload.cpp:20:5: note: suggested alternative: ‘u_int32_t’
     uint32_t offset;
     ^~~~~~~~
     u_int32_t
dfuload.cpp:21:5: error: ‘offset’ was not declared in this scope
     offset =
     ^~~~~~
dfuload.cpp:21:5: note: suggested alternative: ‘off_t’
     offset =
     ^~~~~~
     off_t
dfuload.cpp: In function ‘void fill_clusters(FILE*, int, bpb33*)’:
dfuload.cpp:119:5: error: ‘uint32_t’ was not declared in this scope
     uint32_t clust_size;
     ^~~~~~~~
dfuload.cpp:119:5: note: suggested alternative: ‘u_int32_t’
     uint32_t clust_size;
     ^~~~~~~~
     u_int32_t
dfuload.cpp:120:5: error: ‘uint8_t’ was not declared in this scope
     uint8_t *buf;
     ^~~~~~~
dfuload.cpp:120:5: note: suggested alternative: ‘u_int8_t’
     uint8_t *buf;
     ^~~~~~~
     u_int8_t
dfuload.cpp:120:14: error: ‘buf’ was not declared in this scope
     uint8_t *buf;
              ^~~
dfuload.cpp:121:13: error: expected ‘;’ before ‘rdl’
     uint32_t rdl;
             ^~~~
             ;
dfuload.cpp:122:13: error: expected ‘;’ before ‘total’
     uint32_t total = 0;
             ^~~~~~
             ;
dfuload.cpp:124:5: error: ‘clust_size’ was not declared in this scope
     clust_size = bpb->bpbSecPerClust * bpb->bpbBytesPerSec;
     ^~~~~~~~~~
dfuload.cpp:125:20: error: expected primary-expression before ‘)’ token
     buf = (uint8_t*)malloc(clust_size);
                    ^
dfuload.cpp:130:2: error: ‘rdl’ was not declared in this scope
  rdl = fread(buf, 1, clust_size, fd);
  ^~~
dfuload.cpp:138:9: error: ‘total’ was not declared in this scope
         total += rdl;
         ^~~~~
dfuload.cpp:138:9: note: suggested alternative: ‘strtol’
         total += rdl;
         ^~~~~
         strtol
dfuload.cpp:139:14: error: ‘rdl’ was not declared in this scope
     } while (rdl == clust_size);
              ^~~
dfuload.cpp:141:46: error: ‘total’ was not declared in this scope
     dprintf("\nFinished writing %d bytes\n", total);
                                              ^~~~~
dfuload.cpp:141:46: note: suggested alternative: ‘strtol’
     dprintf("\nFinished writing %d bytes\n", total);
                                              ^~~~~
                                              strtol
[... 190 lines cut ...]
fat12tool.cpp:754:18: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
           return NULL;
                  ^~~~
fat12tool.cpp:768:9: error: ‘set_fat_entry’ was not declared in this scope
         set_fat_entry(i, 0, fd, bpb);
         ^~~~~~~~~~~~~
fat12tool.cpp:768:9: note: suggested alternative: ‘setstate_r’
         set_fat_entry(i, 0, fd, bpb);
         ^~~~~~~~~~~~~
         setstate_r
fat12tool.cpp:773:15: error: ‘uint8_t’ was not declared in this scope
       memset((uint8_t*)&dirent, 0, sizeof(struct direntry));
               ^~~~~~~
fat12tool.cpp:773:15: note: suggested alternative: ‘u_int8_t’
       memset((uint8_t*)&dirent, 0, sizeof(struct direntry));
               ^~~~~~~
               u_int8_t
fat12tool.cpp:773:23: error: expected primary-expression before ‘)’ token
       memset((uint8_t*)&dirent, 0, sizeof(struct direntry));
                       ^
fat12tool.cpp:776:46: error: ‘cluster_to_addr_i’ cannot be used as a function
       int root = cluster_to_addr_i(0, fd, bpb);
                                              ^

```

gabonator, 2019-04-02:
Ok, I tired to build on linux machine and to do so, you need to do these changes:
instead
#include "i386/endian.h"
use
#include "endian.h"

in both fat12tool.cpp and dfuload.cpp before the #include "fat12.h" add this line:
#include <stdint.h>

then the compiler will complain that it cannot convert NULL to integer, so just change those lines from:
return NULL;
to 
return 0;

and it should work

endeavour, 2019-04-02:
Thanks, still getting two errors:
```
dfuload.cpp: In function ‘int main(int, const char**)’:
dfuload.cpp:193:15: error: ‘F_NOCACHE’ was not declared in this scope
     fcntl(fd, F_NOCACHE, 1);
               ^~~~~~~~~
fat12tool.cpp: In function ‘int main(int, const char**)’:
fat12tool.cpp:689:15: error: ‘F_NOCACHE’ was not declared in this scope
     fcntl(fd, F_NOCACHE, 1);
               ^~~~~~~~~

```

gabonator, 2019-04-02:
yes, I forgot. Comment these lines out, I noticed they have no effect anyway

endeavour, 2019-04-02:
Ok great it builds, will try and flash system.hex and let you know how I get on.

endeavour, 2019-04-02:
Do I need to mount the disk? I tried this without any joy:
```
$ ./dfuload /dev/disk/by-label/DFU\\x20V3_61_D cp ../../system/release/system.hex 
Failed to open disk '/dev/disk/by-label/DFU\x20V3_61_D', errno:13

```

gabonator, 2019-04-02:
No, the disk must be not mounted, otherwise you won't get full access to the device. So if you are referring to the disk through volume label, there is high probability that the disk is already mounted.

endeavour, 2019-04-02:
Great, I managed to flash system.hex successfully :) Is there a way to use dfuload to copy the remaining files from approot in one go with wildcards or do I have to send each file individually?

gabonator, 2019-04-02:
just turn the device off and on, mount it and use as regular usb memory stick

endeavour, 2019-04-02:
Fantastic, all working now! Thanks for all your help.

gabonator, 2019-04-02:
great to see that with those little changes it works on linux too. I will try to fix those issues so it will build on both platforms without any problems

gabonator, 2019-04-05:
fixed by this commit
https://github.com/gabonator/LA104/commit/d632b518254a1dfd412abbfb9d439372e5496261

closed 2019-04-05 by gabonator