gabonator/LA104 #5: ds213 compatibility

patrickelectric, 2019-05-17:


gabonator, 2019-07-15:
in progress

gabonator, 2019-09-12:
[image]

The operating system was ported to all devices I own (DS203 with hardware revision 2.60, DS213 and LA104). Currently I am working on simple oscilloscope and signal generator applications for DS203. Unfortunately, the DS213 has requires firmware update from the manufacturer, because the FPGA code has some issues with analog trigger

closed 2019-11-18 by gabonator

blchinezu, 2020-05-09:
It it possible that the FPGA issues could be solved by the FW update released on 12th of March?
http://minidso.com/forum.php?mod=viewthread&tid=3357&extra=page%3D1

I'm actually eager to try your OS. I wonder, can DS213 decently work as a Logic Analyzer or it's better to buy the LA104 too?