gabonator/LA104 #6: Reports file too large

cnorris517, 2019-05-24:
I've tried to deploy this several times to my LA104  DFU v3.61D
Every time I do the copy fails at exactly 66%, the DFU disk disconnects and when it reconnects the system.hex has been renamed system.err. 
If I then power cycle the device I get an image not found error
I did manage to re-flash with firmware from Miniware but i'd really like to work with yours. The same issue occurs for someone else running a custom firmware on the DSO here https://youtu.be/Y0BOBghLsGk
Any ideas

mgazza, 2020-08-25:
Same issue using OSX here.

gabonator, 2020-08-25:
did you use the "dfuload" tool to upload the firmware? https://github.com/gabonator/LA104/tree/master/resources/tutorial_building#flashing-and-first-run

mgazza, 2020-08-25:
I'm not building it. I suspect the OP isn't either. I followed [this](https://github.com/gabonator/LA104#installation) section on the main readme, copied [la014.hex](https://github.com/gabonator/LA104/releases/download/2020.1.mini/la104.hex)

gabonator, 2020-08-25:
I know, but the very last part of that tutorial is about flashing the firmware. The installation notes from the main page are not very accurate. On OSX you need to use the dfuload tool to upload the firmware (at least in my case direct copy did not work)

mgazza, 2020-08-25:
Thanks for the quick replies. I'll give that a go. If it works and I get a chance I'll help updating the readme :D

gabonator, 2020-08-25:
that would be great, my head hurts when I am reading that outdated notes.

mgazza, 2020-08-25:
Notes as I go along. 

## GNU tool chain
can be downloaded from [here] (https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/7-2018-q2-update)

## Inconsistency
path is modified to include the toolchain 
>You will need to change the path to your arm toolchain by changing this line or by exporing the arm toolchain path: export PATH="/Users/gabrielvalky/Downloads/gcc-arm-none-eabi-7-2018-q2-update/bin/":"$PATH"

but later this path is coded into one of the scripts
>Fix the path to arm toolchain in build.sh

## Flashing
cp_la104.sh is located at system/os_host/
if you get the error `Failed to open disk '/dev/disk2', errno:16`
the drive at /Volumes will need to be unmounted.
```
sudo umount /dev/disk2
```

after flashing. turn the device off and on. It should now complain about missing shell.elf. I also had to unplug and re-plugin the usb port for the drive to remount.

mgazza, 2020-08-25:
@gabonator thanks for your help so far. Can't build the shell right now. Any clues?

```
Marks-MacBook-Pro:test29_fileman markgascoyne$ ./build.sh
rm: output.elf: No such file or directory
/Users/markgascoyne/Downloads/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lbios_la104
collect2: error: ld returned 1 exit status
arm-none-eabi-objdump: 'output.elf': No such file
./build.sh: line 11: ../../../../tools/elfstrip/elfstrip: No such file or directory
/Users/markgascoyne/Downloads/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lbios_ds213
collect2: error: ld returned 1 exit status
arm-none-eabi-objdump: 'output.elf': No such file
./build.sh: line 16: ../../../../tools/elfstrip/elfstrip: No such file or directory
/Users/markgascoyne/Downloads/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lbios_ds203
collect2: error: ld returned 1 exit status
arm-none-eabi-objdump: 'output.elf': No such file
./build.sh: line 21: ../../../../tools/elfstrip/elfstrip: No such file or directory
DS203 memory placement:
/Library/Developer/CommandLineTools/usr/bin/nm: error: output.elf: No such file or directory.
Marks-MacBook-Pro:test29_fileman markgascoyne$ ls -la
total 56
drwxr-xr-x  11 markgascoyne  staff   352 Aug 25 16:57 .
drwxr-xr-x   4 markgascoyne  staff   128 Aug 25 15:38 ..
-rw-r--r--   1 markgascoyne  staff  2286 Aug 25 15:38 CMakeLists.txt
-rwxr-xr-x   1 markgascoyne  staff   834 Aug 25 15:38 app.lds
-rwxr-xr-x   1 markgascoyne  staff   885 Aug 25 15:38 app_ds203.lds
drwxr-xr-x   4 markgascoyne  staff   128 Aug 25 17:51 build
-rwxr-xr-x   1 markgascoyne  staff  2178 Aug 25 16:57 build.sh
-rwxr-xr-x   1 markgascoyne  staff    84 Aug 25 15:38 build_mac.sh
-rwxr-xr-x   1 markgascoyne  staff  1616 Aug 25 15:38 build_wasm.sh
-rwxr-xr-x   1 markgascoyne  staff   104 Aug 25 15:38 cp.sh
drwxr-xr-x   5 markgascoyne  staff   160 Aug 25 15:38 source
Marks-MacBook-Pro:test29_fileman markgascoyne$
```

mgazza, 2020-08-25:
Also I think I can simplify your build process by dockerizing the build. once I get everything building...:)

gabonator, 2020-08-25:
you need to build libraries first in system/os_library by running build.sh

gabonator, 2020-08-26:
It seems this discussion provides enough information for flashing LA104. The information provided in readme could be misleading and a flashing guide in separate document should be added. Looking for volunteers :)

closed 2020-08-26 by gabonator