gabonator/LA104 #7: CAN bus analyzing [enhancement]

MstrGearbox, 2019-06-22:
Would it be possible to add automotive CAN bus analyzing or possible decoding?

gabonator, 2019-06-23:
Good idea, actually I planned to make some application with MCP2515 canbus transceiver module attached to LA104, so instead of using logic probe pins, all the canbus traffic would go through MCP2515 chip which allows not only analysis of the traffic, but also allows to generate new packets. But I don't know when I will start doing some research on this topic, I was hoping that other people will start contributing to this project with their own applications and one of it could be this CAN analyser :)

jilleb, 2022-08-19:
I got some LIN-UART transceivers on the way.. I might make a nice LIN analyzer for the LA104 😎