gabonator/LA104 #8: DS203 "Device identification failed"

danielkucera, 2019-12-16:
Hi,

I'm trying to upload system.hex to my DS203 but it fails:
```
~/Projects/LA104/tools/dfuload$ sudo ./dfuload /dev/sdc cp ~/Downloads/system.hex 
OemName: MSDOS5.0
Boot boot sector signature 00
Device identification failed
Closing disk Ok
```
dmesg:
```
[52453.861781] usb 2-3: new full-speed USB device number 11 using xhci_hcd
[52454.014977] usb 2-3: New USB device found, idVendor=0483, idProduct=5720
[52454.014981] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52454.014983] usb 2-3: Product: STM32 Mass Storage
[52454.014985] usb 2-3: Manufacturer: STMicroelectronics
[52454.014986] usb 2-3: SerialNumber: 8ÿx33T457YC
[52454.015738] usb-storage 2-3:1.0: USB Mass Storage device detected
[52454.016296] scsi host3: usb-storage 2-3:1.0
[52455.030484] scsi 3:0:0:0: Direct-Access     Vertual  DFU Disk              PQ: 0 ANSI: 2
[52455.031546] sd 3:0:0:0: Attached scsi generic sg2 type 0
[52455.031738] sd 3:0:0:0: [sdc] 1024 512-byte logical blocks: (524 kB/512 KiB)
[52455.031881] sd 3:0:0:0: [sdc] Write Protect is off
[52455.031883] sd 3:0:0:0: [sdc] Mode Sense: 03 00 00 00
[52455.032023] sd 3:0:0:0: [sdc] No Caching mode page found
[52455.032027] sd 3:0:0:0: [sdc] Assuming drive cache: write through
[52455.051395] sd 3:0:0:0: [sdc] Attached SCSI removable disk
```
dd of my device:
[orig-system.zip](https://github.com/gabonator/LA104/files/3967033/orig-system.zip)

danielkucera, 2019-12-16:
So I did this:
```
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sudo mount /dev/sdc /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sudo cp ~/Downloads/system.hex /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sudo umount /mnt/tst
```
and after a restart I got white screen and:
```
[53532.958590] usb 2-3: new full-speed USB device number 73 using xhci_hcd
[53543.426569] usb 2-3: device descriptor read/64, error -110
[53543.666589] usb 2-3: device descriptor read/64, error -71
[53543.902667] usb 2-3: new full-speed USB device number 74 using xhci_hcd
```

gabonator, 2019-12-16:
Try this hex file:
[ds203.hex.txt](https://github.com/gabonator/LA104/files/3967175/ds203.hex.txt)

After copying the firmware image, the device should disconnect from USB and reconnect and a original file with suffix ".RDY" should appear if the file was successfully flashed. Do you see that file?

danielkucera, 2019-12-16:
No progress:

```
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sudo mount /dev/sdc /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sudo cp ~/Downloads/ds203.hex.txt /mnt/tst/ds203.hex
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
ds203.hex
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
ds203.hex
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
ds203.hex
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
ds203.hex
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
ds203.hex
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sync
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ 
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ ls /mnt/tst
danman@silverhorse:~/Projects/LA104/tools/dfuload$ sudo umount /mnt/tst
```

gabonator, 2019-12-16:
find me on facebook: Archangel Gab

danielkucera, 2019-12-16:
I've sent you a message. It might be in spam.

closed 2019-12-21 by danielkucera

gabonator, 2019-12-22:
To sum up what we learned:

It was impossible to flash new firmware using DFU V3.11C, the device was not properly recognized or the file transfer was failing. We had to downgrade to DFU V3.10 using internal UART with stm32flash utility:
https://github.com/gabonator/LA104/tree/master/tools/stm32flash

For DS203 you need special image, not the one provided in release section. (build_ds203.sh in https://github.com/gabonator/LA104/tree/master/system/os_host)

When the OS was running, it was failing to run shell.elf. It just said "Image loaded at 0x0801f000" but the device. Building the image for different addres solved the problem (entry point moved to 0x08019000)

Running WebUsb (68webdrv.elf) - you need to reconnect the USB cable after running this app. Chrome notification should appear, but the communication on Ubuntu 18.0.4 did not work (Unable to claim interface in developer console). Problem was resolved by following command: "sudo rmmod cdc_acm" to release the system modem device driver.