gabonator/LA104 #9: system_ds203.hex not loading on DS203 (hardware 2.70)

fredzo, 2019-12-26:
Hi Gabriel,

I wanted to try your new firmware on my good old DS203 but I can't get it to load (DFU reports .rdy file but after reboot I still have the previously loaded firmware).
My DS203 has a 2.70 hardware, could this be the problem ?
I managed to load successfully your webusbosc firmware tough (great job by the way !).

Thanks four your help !

Frederic.

gabonator, 2019-12-26:
Which firmware did you try loading?
This is the right one (ds203.hex):
https://github.com/gabonator/LA104/tree/master/system/apps_ds213/69_webusbosc/release

fredzo, 2019-12-26:
Oh yes, this one worked perfectly and I was able to run the 69webosc app, congratulations ! ;-)
But in this version there is no graphical app chooser and no other apps right ?
I was hopping that this one https://github.com/gabonator/LA104/blob/master/system/release/system_ds203.hex was a complete version of the ds203 firmware with other applications, but maybe you're not quite there yet on the port to ds203 ;-)
Please let me know if you have plans on going on with this port and if I can help with that !
Do you have some instructions on how to setup the build chain, ideally on a Windows environment ?
I had it working back in 2013 when working on the first version of your ds203 firmware, but you don't seem to be using VisualStudio anymore right ?

gabonator, 2019-12-26:
This time it works a bit different - instead of one single huge firmware file, the firmware is only a simple operating system which loads "elf applications". This system exposes all necessary functions to the application layer. Applications are copied onto the internal SPI flash, so the development is much faster and also it allows you to have tons of different apps on flash and switch between them. This system was primarily designed for logic analyzer LA104, and for this device I made about 30 useful applications. For the DS203 I have only ported my old oscillosocope firmware just to verify if everything works well. Maybe I will try to port also other popular alternative firmwares (e.g. MotoMaxis DS203), it would be great to have all of the alternative/official firmwares inside one device. But currently I am focused on improving the WebUsb web interface. This offers much more possibilities of decoding and visualization.

The toolchain I am using is this one (there are win32 builds as well):
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

For building the os, see [system/os_host/build_ds203.sh](../tree/master/system/os_host/build_ds203.sh), it could be easily rewritten into windows batch file.

The simplest application example is in [system/apps_experiments/test2_import/](../tree/master/system/apps_experiments/test2_import/)
again, you need to rewrite the build script build.sh into batch file (ignore the calls to elfdump)

ELF images usually contains some useless parts, and they needs to be removed to achieve even smaller resulting application image files. For this purpose there is [tools/elfstrip/main.cpp](../tree/master/tools/elfstrip/main.cpp) which you need to compile (this should not be a problem, this program was initially made for win32)

I don't use visual studio, but for some larger projects there are also cmake files which you can use to generate msvc project files. I use it for debugging applications in xcode and building as native desktop apps, it requires SDL2 library. If you want to try it, here it is: [apps_ds213/55_gabosc](../tree/master/system/apps_ds213/55_gabosc), build_mac.sh generates the xcode project files, try changing the commandline to generate MSVC project. For small projects I use just text editor with syntax highlighting

If you want to contribute I recommend to make some improvemens in the web interface for the webusb oscilloscope. But it is completely up to you :) Applications for DS203 (or DS213) are in [system/apps_ds213](../tree/master/system/apps_ds213) folder

fredzo, 2019-12-27:
Great, thanks for the detailed answer Gabriel !
I'll definitely take a look into that, and most likely come back to you with some questions ;-)

closed 2020-01-11 by gabonator