gabonator/LA104 #12: Unable to load new firmware

finale-z, 2020-02-15:
I ran into the same situation as the previous issue. I couldn't load the new firmware on v3.61d. I also successfully updated the official LA104_APP_V1.03.hex, but still couldn't update the latest la104.hex. I tried to use dfuload, but only returned：
oemname: msdos5.0
Nothing to do.
Closing disk Ok

finale-z, 2020-02-15:
At first, la104.hex will return la104.err after adding la104.hex, but now he seems to ignore the firmware I copied in completely. After the firmware is uploaded, the disk is disconnected and reconnected, but la104.hex remains the same

gabonator, 2020-02-15:
for the first comment, run the dfuload with these arguments:
sudo ../../tools/dfuload/dfuload /dev/disk2 cp ./build/system_la104.hex
the first argument is the target device, the second is "cp" and third is path to the file you want to flash. Script which does this is located in /system/os_host/cp_la104.sh
you just need to change the device name and it should work

finale-z, 2020-02-15:
I found a simple and crude solution, using dd to forcibly write hex files under Linux.
Thank you for your answer

finale-z, 2020-02-15:
But there is a problem with this method. After using dd to burn hex files and power on again, Linux can't find the disk, so it can only copy elf files under windows

gabonator, 2020-02-15:
so did you manage to upload the firmware to DFU? if not, try the hex file from this release: https://github.com/gabonator/LA104/releases/tag/2020.1.mini

finale-z, 2020-02-15:
I tried to upload this version of firmware to DFU, but as I said earlier, it didn't work. Then I tried to burn the firmware to DFU using dd, and it worked

closed 2020-02-15 by finale-z