gabonator/LA104 #13: Cannot get PulseView to load

bl1tz3n4, 2020-02-24:
Initially had issues copying system.hex on windows 10 and Ubuntu, could not get DFUload to open. Tried mini installation mentioned in another issue and that loaded, but PulseView will not open.

Reports "Symbol '_ZN4BIOS3USB10InitializeEPvS1_S1S1_PPFvvES4_S4_' not found!" 

And 

"Assertion failed in ../source/main/Execute.cpp [432]: dwProcAddr
Application halted"

I presume this is because of the mini installation, but can't seem to get the "full" installation to install

gabonator, 2020-02-24:
Here is a fresh rebuild of 32sump application:
[32sump.elf.zip](https://github.com/gabonator/LA104/files/4243534/32sump.elf.zip)

But do not expect much, it is just proof of concept, the sampling frequency is fixed at 10kbps and it samples the P1..P4 pins

gbraad, 2020-03-14:
@gabonator I can get pulseview to find the device, but the sample does not return anything...

nieldk, 2021-03-17:
> Initially had issues copying system.hex on windows 10 and Ubuntu, could not get DFUload to open. Tried mini installation mentioned in another issue and that loaded, but PulseView will not open. [...]

I suggest trying to reinstall, but before unzipping the apps to the LA104, format the disk.