gabonator/LA104 #22: Add license information

vazhnov, 2020-10-01:
Hello!

I found link to this project from https://hackaday.com/2020/10/01/teaching-a-pocket-logic-analyzer-many-new-tricks/

Could you please add information about license of this repository — is it allowed to copy, change and use code in any purpose?
It would be great to see any popular open source license.

gabonator, 2020-10-01:
Hello, I am completely uneducated in this field. Can you please recommend me any specific license for this? I provide this for the community and everyone is free to do whathever he wants to do with it. Many components are taken from other projects, and in most case I left there credit for them (but didn't study their license), so I don't know how to proceed correctly in this case

superhero1, 2020-10-02:
Maybe this can help: https://choosealicense.com/

gabonator, 2020-10-02:
@superhero1 Thank you, MIT license added in this PR
https://github.com/gabonator/LA104/pull/23

closed 2020-10-02 by gabonator