gabonator/LA104 #24: Sequencer doesn't work in lateste Beta

wbvreeuwijk, 2020-10-21:
Hi,

First of all, thanks for creating this project. It makes these devices even more useful than they already are.
I recently installed the beta release, and most of the apps work; however, the sequencer app appears not to accept the SMPLbutton on the menu. So I'm not able to get into settings or select Loop, Burst, or Follow.

Since it is a beta release, this may already be on your radar; if so, then please ignore this message :-)

gabonator, 2020-10-21:
Thank you very much for testing and reporting this issue. I fixed it in this commit: https://github.com/gabonator/LA104/commit/cba00e61ee7843a6dc025f4df2b010d4377f3306

you can download fixed version here:
https://github.com/gabonator/LA104/blob/master/system/release/bin/22seqen.elf

wbvreeuwijk, 2020-10-23:
Thanks.

I'm now able to select the Loop and Follow functions. The settings and File options don't work for me yet however. Als when trying to exit the application I get the Press F1 to retry message and 'shell.elf' not found error.

gabonator, 2020-10-25:
Please try this update:
[sequencer_update2.zip](https://github.com/gabonator/LA104/files/5434343/sequencer_update2.zip)

You will need two PCF8574 I2C expanders to use this app, otherwise it will work in offline mode. I added there notification messages and removed unimplemented menu tabs

wbvreeuwijk, 2020-10-26:
Thanks.

This is working. I tested with two PCF8574's I had laying around. One thing that may be useful to add to these apps is how to connect and what addresses to set the PCF8574 to. I had to look at the code to figure this out. Maybe a short Readme.md.
Another useful feature would be to be able to set the speed.

closed 2020-10-26 by wbvreeuwijk

gabonator, 2020-10-27:
Added short manual for those applications:
[image]
[image]
[image]