gabonator/LA104 #27: Explaingin the Repository structure

Markus-Be, 2020-11-01:
Hi,

iam trying to understand and document some details in this repository at the moment.
Can someone please explain the structure of that repository?

If i run the build scripts (for example in root folder or "build_all_apps.sh" in system) i get errors about missing files.

How to get started?

gabonator, 2020-11-01:
Does this help? 

https://github.com/gabonator/LA104/tree/master/resources/tutorial_building

I think the only thing you need to install before building is nodejs and arm toolchain 
(https://developer.arm.com/-/media/Files/downloads/gnu-rm/7-2018q2/gcc-arm-none-eabi-7-2018-q2-update-linux.tar.bz2)

What kind of errors you are getting?

gabonator, 2020-11-02:
I have recently changed building process. See the system/scripts/build.sh script. This script is also used for docker automated builds and should run on any linux machine after installing the arm toolchain, nodejs, zip and imagemagick

gabonator, 2020-11-27:
added repo structure: https://github.com/gabonator/LA104#repository-structure

closed 2020-11-27 by gabonator