gabonator/LA104 #28: DS203 missing shell.elf

Compycompy, 2020-11-04:
Hello, 
I have a DS203 currently with wildcat 6.5 hw 2.81 sys 1.64
I went in DSU mode and copied the ds203.hex file, all went fine.
Disconnected, restarted and asked for shell.elf, I copied the elf files but still keeps on asking for the shell.elf file.
I believe there is some issue with the memory location as it seems to look in another part but I am not sure.
Any suggestion?

gabonator, 2020-11-04:
Try this firmware:
[system_ds203.hex.zip](https://github.com/gabonator/LA104/files/5487535/system_ds203.hex.zip)

And copy these files to devices:
[ds203apps.zip](https://github.com/gabonator/LA104/files/5487536/ds203apps.zip)

If it won't work, please send here screenshot of the root of the device's filesystem

Compycompy, 2020-11-04:
Hi gab,
 I tried to flash clean with the RS232 port, all was smooth, I copied the sys_b164.hex and all started correctly.
I loaded the system_ds203.hex, it was accepted. Turned off, then on asking for shell.elf (which at this point is fine). 
I copied the ds203apps (uncompressed), turned off then on but still asking for shell.elf    :-( 
On the device directory I have the seven elf files, CALIB.DAT and CONFIG.DAT
Thanks for the help

gabonator, 2020-11-04:
I just tried the files I provided and with the same result as in your case. For some strange reason, the filesystem gets corrupted when I copied all the files at once. After reconnecting I saw only empty root folder. But when I copied only the shell.elf to the disk, it worked. And then when the shell was loaded I copied remaining elf files. So please try to delete all *.elf files you copied before and copy only "shell.elf". Or try copying the file until the device will find it, possibly device format could help (dont forget to backup your CALIB.DAT and CONFIG.DAT)

Compycompy, 2020-11-05:
Unfortunately no luck always missing shell.elf (tried to format both quick and standard). I tried either pressing f1 directly after copying or doing power cycle (several times).
I will try with another PC/OS maybe it helps. 
Thank you!

gabonator, 2020-11-05:
Just out of curiosity, please drop here screenshot of the disk root and hex view of the shell.elf

Compycompy, 2020-11-05:
Hi, 
this is the shell.elf hex view and the folder.
[image]

[image]

thank you!

gabonator, 2020-11-05:
try running "fsutil fsinfo ntfsinfo c:", this should show us how the disk is formatted, the disk should have 4096 sectors and every sector should have 512 bytes, on OSX:
```
$ sudo diskutil info /dev/disk2                                      ↑
   Device Identifier:         disk2
   Device Node:               /dev/disk2
   Whole:                     Yes
   Part of Whole:             disk2
   Device / Media Name:       Flash Disk

   Volume Name:               DS203
   Mounted:                   No

   Content (IOContent):       None
   File System Personality:   MS-DOS FAT12
   Type (Bundle):             msdos
   Name (User Visible):       MS-DOS (FAT12)

   OS Can Be Installed:       No
   Media Type:                Generic
   Protocol:                  USB
   SMART Status:              Not Supported
   Volume UUID:               FC397130-DDE8-3700-94A9-F8B250F2079E

   Disk Size:                 2.1 MB (2097152 Bytes) (exactly 4096 512-Byte-Units)
   Device Block Size:         512 Bytes

   Volume Total Space:        0 B (0 Bytes) (exactly 0 512-Byte-Units)
   Volume Free Space:         0 B (0 Bytes) (exactly 0 512-Byte-Units)

   Read-Only Media:           No
   Read-Only Volume:          Not applicable (not mounted)

   Device Location:           External
   Removable Media:           Removable
   Media Removal:             Software-Activated

   Solid State:               Info not available
   Virtual:                   No
```

Compycompy, 2020-11-05:
Hello gab, 
 my disk size is 8mb not 2, could that be a problem?
I try to run the command later today from linux.
thank you

gabonator, 2020-11-05:
Yes that is a problem, the disk parameters are hard coded in the firmware. I do not own DS203 with 8MB disk, but if you can tell me what is the sector size and how many sectors are in that disk, I can prepare a build for you (and for others who own this newer version of DSO203), on linux try this command "fdisk -l"

Compycompy, 2020-11-05:
From what I can see from windows it is exactly 8.388.608 byte with 4096 byte/sector and 2048 total sectors.
I can look with the command under linux if needed.

gabonator, 2020-11-05:
Try this
[system_ds203hd.hex.zip](https://github.com/gabonator/LA104/files/5494951/system_ds203hd.hex.zip)
[ds203apps.zip](https://github.com/gabonator/LA104/files/5494952/ds203apps.zip)

Compycompy, 2020-11-05:
Copied in DFU mode the new system, now it goes directly in file manager mode.
Reboot and in normal mode I copied the shell.elf
Reboot and goes into file manager, if I choose the shell.elf some text appears quickly at the bottom (cannot see what is coming up) and then goes back to the file manager like if nothing happened.
I copied all the other elf files, they all work with the exception of miniosc.elf which gives an hardware error.

[image]

gabonator, 2020-11-05:
Try one of these, maybe it will work. Their (minidso's) bios use different memory layout, because it needs bigger buffer for the filesystem (4k vs. 512), so I tried to build the app for different memory locations:
[Archive.zip](https://github.com/gabonator/LA104/files/5495533/Archive.zip)

Compycompy, 2020-11-05:
No luck always hardware fault :-(
Btw I noticed that since I flashed the new system file it quickly says at boot FPGA configuration error, but then it continues without problems.

gabonator, 2020-11-05:
Without having access to your device it is very difficult to fix this issue. And when it says that there is problem with FPGA, then also the oscilloscope functions wont work even when the application was running.

Compycompy, 2020-11-09:
Ok thanks a lot anyway. The FPGA error has been resolved with the recovery with serial. Do you know if I can copy the old version of your system on the 8m version? (the GAB_N1.hex, N2, N3 and N4)

acidg, 2020-11-13:
I did quite the same as @Compycompy and ended up with the FPGA configuration error and corrupted logo. I have HW version 2.72. Any chance to get this working or revert to the old version?

acidg, 2020-11-18:
@Compycompy can you tell me how you did the recovery with serial? Where did you get the binaries for the FPGA?

gabonator, 2020-12-09:
Hello guys, do any of you have working 8MB device with original fw? Could you please share flash image of your device? I am trying to put together all flash images for recovery purposes. Thank you! (fyi: @nutz88)

Sergio-, 2020-12-15:
Hi.
I wish I saw this earlier. I had a DSO203 8Mb working until few minutes ago. I tried to updated the firmware and now I'm stuck on the "Image 'shell.elf' not found" message. :-(

acidg, 2020-12-15:
You can find the original fw here: http://jpa.kapsi.fi/dsoquad/ as DS203.V2.72.zip

Sergio-, 2020-12-16:
Forgot to mention: My Hardware version is 2.82, Sys ver 1.64.
Is there firmware out there I can use to put the device back to work? I tried putting the GAB_N1.hex into the DFU disk, but only got the ERR file back. I'm using Ubuntu 20.04 as Environment.
Thank you.

acidg, 2020-12-16:
You can find Firmware 2.81 with FPGA binaries here: http://minidso.com/forum.php?mod=viewthread&tid=666&extra=page%3D1
I repacked the files for you: [DS203_2.81_HW2.82.zip](https://github.com/gabonator/LA104/files/5701407/DS203_2.81_HW2.82.zip)

Sergio-, 2020-12-17:
> You can find Firmware 2.81 with FPGA binaries for HW version >= 2.82 here: http://minidso.com/forum.php?mod=viewthread&tid=666&extra=page%3D1
> I repacked the files for you: [DS203_2.81_HW2.82.zip](https://github.com/gabonator/LA104/files/5701407/DS203_2.81_HW2.82.zip)

Hello @acidg Thanks so much for putting this together. Really appreciate.
All went well, except the last file PA1_V113.hex. I tried putting it in several times, it always return as ERR. I tried the dfuload method, but I was not able to put the device back to work. It returning FPGA configuration error:
Do you guys think this is permanent damaged or do we still have ways to recover it?
Thank you.
[image]

Sergio-, 2020-12-17:
> Ok thanks a lot anyway. The FPGA error has been resolved with the recovery with serial. Do you know if I can copy the old version of your system on the 8m version? (the GAB_N1.hex, N2, N3 and N4)

Hi @Compycompy Could you please point us or share how to proceed with the serial recovery to fix the FPGA issue? Thank you.

gabonator, 2020-12-17:
@Sergio- it is nearly impossible to brick the device, DS203 can be recovered using the internal uart connection: https://github.com/gabonator/LA104/tree/master/tools/stm32flash
You will need uart converter (3.3V) for this

Sergio-, 2020-12-17:
> @Sergio- it is nearly impossible to brick the device, DS203 can be recovered using the internal uart connection: https://github.com/gabonator/LA104/tree/master/tools/stm32flash
> You will need uart converter (3.3V) for this

Hi @gabonator! First of all, thanks for the reply. 
It's a good relief to know that the device can be restored. I believe it's just a matter to find the correct firmware combination now. :-)

I'll explain below what I did to recover from UART, hoping that others can user the information too.
Before starting, I took a picture of my device first screen, where it shows the license (I'm not sure if that's useful or not).
I compiled a Linux version of the stm32flash utility (download from sources, build it replace the bin at the same location I had the one from your repository).
I removed the cover and battery of the DSO203, and soldered some wires on the BOOT0 and UART ports (I didn't have any pin bars available) :-)
[image]
[image]
I connected the RX, TX and GND pins into USB to Serial adapter (3.3V) and to enter the Bootloader mode, just jumped the B0 with VCC, then powered on the device (the screen was just white at this point).
Then I run:
`sudo ./stm32flash -w ./ds203_2mb_full/dso_quad_bootloader_flash_modified.bin /dev/ttyUSB0 `
and got this output
```
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0414 (STM32F10xxx High-density)
- RAM        : 64KiB  (512b reserved by bootloader)
- Flash      : 512KiB (size first sector: 2x2048)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote address 0x08040000 (100.00%) Done.
```
I then disconnected the B0 from VCC, powered off the device and on again. 
I got an Invalid License screen:
[image]
Even inputting the number I recorder earlier, it didn't work. Anyway, you can press the top keys to move to the next steps.
Once the main application is loaded. I noticed something strange, the 2 channels (CHA, CHB) are not displaying. Not sure how to solve that.
The other issue is that, when booting pressing F1, the DFU disk is not being mounted anymore. The logs shows the message below:

```
106093.808621] usb 1-3: new full-speed USB device number 61 using xhci_hcd
[106093.957735] usb 1-3: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[106093.957738] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[106093.957740] usb 1-3: Product: STM32 Mass Storage
[106093.957743] usb 1-3: Manufacturer: STMicroelectronics
[106093.957744] usb 1-3: SerialNumber: 6ÿÛMW60 cQ
[106093.958882] usb-storage 1-3:1.0: USB Mass Storage device detected
[106093.959063] scsi host2: usb-storage 1-3:1.0
[106094.973067] scsi 2:0:0:0: Direct-Access     Mini_DSO Flash Disk            PQ: 0 ANSI: 2
[106094.973316] sd 2:0:0:0: Attached scsi generic sg1 type 0
[106094.973576] sd 2:0:0:0: [sdb] 4096 512-byte logical blocks: (2.10 MB/2.00 MiB)
[106094.973720] sd 2:0:0:0: [sdb] Write Protect is off
[106094.973724] sd 2:0:0:0: [sdb] Mode Sense: 03 00 00 00
[106094.973865] sd 2:0:0:0: [sdb] No Caching mode page found
[106094.973869] sd 2:0:0:0: [sdb] Assuming drive cache: write through
[106095.045694] sd 2:0:0:0: [sdb] Attached SCSI removable disk
[106095.988819] sd 2:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
[106095.988823] sd 2:0:0:0: [sdb] tag#0 CDB: Read(10) 28 00 00 00 0d 58 00 00 08 00
[106095.988827] blk_update_request: I/O error, dev sdb, sector 3416 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[106095.988832] usb 1-3: USB disconnect, device number 61
[106095.988895] blk_update_request: I/O error, dev sdb, sector 3416 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[106095.988898] Buffer I/O error on dev sdb, logical block 427, async page read
[106096.888682] usb 1-3: new full-speed USB device number 62 using xhci_hcd
[106097.037671] usb 1-3: New USB device found, idVendor=0483, idProduct=5720, bcdDevice= 2.00
[106097.037673] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[106097.037674] usb 1-3: Product: STM32 Mass Storage
[106097.037675] usb 1-3: Manufacturer: STMicroelectronics
[106097.037676] usb 1-3: SerialNumber: 6ÿÛMW60 cQ
[106097.038736] usb-storage 1-3:1.0: USB Mass Storage device detected
[106097.038840] scsi host2: usb-storage 1-3:1.0
[106098.045137] scsi 2:0:0:0: Direct-Access     Vertual  DFU Disk              PQ: 0 ANSI: 2
[106098.045408] sd 2:0:0:0: Attached scsi generic sg1 type 0
[106098.046502] sd 2:0:0:0: [sdb] 1024 512-byte logical blocks: (524 kB/512 KiB)
[106098.046644] sd 2:0:0:0: [sdb] Write Protect is off
[106098.046647] sd 2:0:0:0: [sdb] Mode Sense: 03 00 00 00
[106098.046789] sd 2:0:0:0: [sdb] No Caching mode page found
[106098.046793] sd 2:0:0:0: [sdb] Assuming drive cache: write through
[106098.088458] sd 2:0:0:0: [sdb] Attached SCSI removable disk

```
To summarize: The device nows bootup, but it seems it's not working as intended still. 
I'm looking for any advice on how to proceed from here. 
Thanks a lot.

gabonator, 2020-12-17:
@Sergio- During powerup, what does it say about FPGA configuration? When you connect CH1 to signal generator, does it show any signal?

If you are using FW from jpa, he made there a little hack that skips license verification, with original fw it wouldn't let you go further.

If you are having problems with DFU drive, you can try uploading new firmwares (HEX files) through this utility https://github.com/gabonator/LA104/tree/master/tools/dfuload

It seems you just need to find the right fw for your device. Or maybe just fpga image. Maybe we could ask guys here for a help: Anyone here with working DS203 V2.82 8MB and willing to share its flash dump with us?

Sergio-, 2020-12-18:
> @Sergio- During powerup, what does it say about FPGA configuration? When you connect CH1 to signal generator, does it show any signal?
>
> If you are using FW from jpa, he made there a little hack that [...]

Hi @gabonator Thanks for the reply!
1 - No it doesn't show any FPGA error now. It jumps directly to that license screen where it shows 0000000.I even recorded it in slow motion to make sure I'm losing any quick message. It jumps directly to the license screen.
2 - Connecting CHA or CHB to a generator doesn't show up anything, changing the channel parameters have not effect.
3 - FW from jpa? I'm loaded via USART the one that one you here on the repository (dso_quad_bootloader_flash_modified.bin) is there any other one that I should consider? I found this repository (http://jpa.kapsi.fi/dsoquad/) but, as I don't have the DFU disk mounting anymore, I don't know how I can load these...
4 - I tried the dfuload method. It seems to work sometimes, but I'm not 100% sure as I cannot read the output file response (ERR, RDY, etc). I tried loading a couple of files, event a FPGA firmware available on the repository here (https://github.com/gabonator/LA104/tree/master/tools/fpga/ds203_8mb_hwV272-V261FPGA), but, still not luck. Devices boots up on the license screen, i press f1 twice to skip it, it load the oscilloscope app, but CHA and CHB doesn't show up. DFU disk is also not mounted when boot holding F1.

gabonator, 2020-12-18:
@Sergio- 
then it explains everything - the dso_quad_bootloader_flash_modified.bin is from jpa (http://jpa.kapsi.fi/dsoquad/) and the "modified" means, he skipped the license verification. The problem is, this flash dump is for 2MB device. So that is probably the reason why you are unable to mount it. Just continue try to find proper fw, and if you won't find anything, I will try to ask the guys at miniware if they can help. But I am not giving much hope they will be able to help us

Sergio-, 2020-12-18:
@gabonator Thanks. I"ll continue searching. 
Let's hope someone with a working 8mb version can post the flash dump here.
Thanks a lot.

Sergio-, 2020-12-18:
[image]

acidg, 2020-12-18:
> [106098.046502] sd 2:0:0:0: [sdb] 1024 512-byte logical blocks: (524 kB/512 KiB)
> [106098.046644] sd 2:0:0:0: [sdb] Write Protect is off
> [106098.046647] sd 2:0:0:0: [sdb] Mode Sense: 03 00 00 00 [...]

As far as I see from this log, your DFU device is there, it's just not mounted automatically. Try to mount the device manually: `sudo mount -o rw /dev/sdb /media/USB`. You should be able to copy the firmware files there. You will have to unmount and mount the device again to see the `.RDY` and `.ERR` files.

Sergio-, 2020-12-18:
> As far as I see from this log, your DFU device is there, it's just not mounted automatically. Try to mount the device manually: `sudo mount -o rw /dev/sdb /media/USB`. You should be able to copy the [...]
Hi @acidg Thanks for the suggestion!
I tried that, althought I can mount the device, it only return "write error" when I try to saving something on it. Even the dfuload utility returns only writing errors...

Sergio-, 2020-12-18:
I was able to make a little progress. I found the attached file on the minidso Chinese forum, it contains the DFU 3.45C for the Hw v2.81. I flashed this file using the stm32flash and it helped a little.
After flashing and rebooting, I got this screen:
[image]
Then, It moves to this one:
[image]
and finally:
[image]

Booting the device holding F1, shows this screen:
[image]
And it mounts the DFU drive on my Linux machine.

But, not everything is Ok yet... When I try to do any operation with the virtual DFU disk, it always return as ERR. If I drag a file and drop on the disk, it immediately disconnects and, upon returning, it contains the file with extension .ERR.
I also tried transferring the files using the 'dfuload' utility. I umount (sudo umount /dev/sdb) then transfer the file, the drive disconnects and, when connecting it always have the A.ERR file in it.

I'm clueless how to move on from here. I want to install a decent firmware like the one from this repo, but, transferring the file via DFU vitual disk or dfuload is not working. I tried upgrading the APP and SYS via same method, but got the same result (ERR). I even tried the Wildcat 6.x method with the HEX files split in BIN. As soon as I save the first ADR file, the drive disconnects (expected), but returns with a ERR file instead of a SET.

Another thing: The attached file also contain the DFU 3.40C. I tried loading it with stm32flash the same way I did with the 3.45 but it always stopped at 40%, no error message, nothing. just stopped at 40%... 

Any ideas/suggestions are appreciated.

[dso_hw281_dfu_pack.zip](https://github.com/gabonator/LA104/files/5717109/dso_hw281_dfu_pack.zip)

Thank you.

MotoMaxis, 2020-12-18:
Sergio, it looks like you have used the wrong image for recovery the first time around. But don't worry, we'll get you up and running.  You just need the correct DPU, FPGA, LOGO, SYS and APP reflashed.

It says hw 2.82 on your PCB and the boot screen says hw 2.60. It means the DPU is wrong too that uses slightly different license check code which explains why your old code didn't work. Now You will need to flash the Unlock app and enter the license to fix the license error.

Starting with hw 2.81 they have changed the memory layout which means older software can't be used with 2.8x as it will corrupt things. 

I happen to have hw 2.82 device and did a rom dump using Wildcat's firmware.  My device has the DPU 3.46C, Wildcats custom FPGA with full speed oversampling, and his software v6.5.  Yes it's far from stock but it should put things where they belong.

I have included the dump and the Unlock app.  First flash the dump. Then see if DPU is working and mounts then load the Unlock app. Reboot and try entering the license. (DFU = 3.46C ser: 8726B9A3 lic: 7A9457A7)

Then load the Wildcat 6.5 app WC65app1.hex.

Good luck.
[HW_V282_DFU_V346C_WC_V65.zip](https://github.com/gabonator/LA104/files/5717537/HW_V282_DFU_V346C_WC_V65.zip)

Sergio-, 2020-12-20:
> Sergio, it looks like you have used the wrong image for recovery the first time around. But don't worry, we'll get you up and running. You just need the correct DFU, FPGA, LOGO, SYS and APP reflashed.
> ...

Hello @MotoMaxis 
First of all, thanks for such a detailed response and Thanks for the commitment in helping. I really appreciate it and I'm sure others will find it useful too.
I flashed the HW_V282_DFU_V346C_WC_V65.BIN using the stm32flash utility. It worked fine and when rebooted, I entered the DFU mode and transferred the UNLOCK.hex (via dfuload).

Upon rebooting, the screen to enter the new license showed up, I input the license and pressed Key2. This is what showed up:
[image]

I turned the device Off and On again, then this screen comes up:
[image]

Then the screen to enter the license comes up again.
[image]

I tried entering and saving the license a couple of times, it always display "Licence written Ok", but comes up back to "License Error" -> "License Unlock" screen.

I re-flashed the DFU and repeated all the steps few times.. No luck...

Any suggestions is appreciated.

Thank you.

MotoMaxis, 2020-12-20:
Once it accepts the License, reboot into DFU and load WC65app1.hex next, only after this do a normal boot.

MotoMaxis, 2020-12-20:
The Unlock is actually an app only for entering the License at times when the device was wiped.

Sergio-, 2020-12-21:
Hi @MotoMaxis Thanks for the quick reply.
I tried that and I'm still with no luck.
I recorded the video below trying to capture all the details I'm doing expecting that something wrong can be identified and pointed to me so I can fix it.
Feel free to skip the obvious sequential parts.
https://drive.google.com/file/d/12sNB2ID2U8ckqJI-EkF01A1hJnLruxs4/view?usp=sharing
I appreciate all the help.
Thank you.

MotoMaxis, 2020-12-21:
[DS203.Unlock.V0.21.hex.zip](https://github.com/gabonator/LA104/files/5722770/DS203.Unlock.V0.21.hex.zip)

Sergio-, 2020-12-21:
Cool @MotoMaxis 
Yes. I tried the new Unlock, but the behavior remained the same as the one in the video.
I'm not that familiar with the conversion of the HEX to BIN and how to place it in the right address, but It seems a good approach, provided we have the address/location and the proper file. 
Is there anything I can do that will help in this process?
Thank you!

MotoMaxis, 2020-12-21:
Finally had time put together the recovery package.
The memory layout looks like this.
```
Only DS203 Hardware Ver 2.81 and 2.82
08000000 - DFU
08004000 - SYS / Unlock
0800C000 - APP
0805C800 - FPGA
0807D800 - LOGO
```
The device must have all 5 sections inplace to function properly. On Power on, operation starts at DFU, then jump to SYS then to APP.  Once you use Unlock, you must write back the SYS,  

I have included the stock dump in IMAGE in hex and bin. Use UART to write it to device. I also included Wildcat 6.5 app to try.
[RECOVERY281282.zip](https://github.com/gabonator/LA104/files/5726823/RECOVERY281282.zip)

Since DFU is so unreliable for you I suggest you use stm32flash utility.  I believe you can "stm32flash -w filename -v -S  0x4000 ..." to write SYS as an example as it lives at 08004000 .  This way you can write what you need.

Best regards.

MotoMaxis, 2020-12-22:
Hey Sergio.  Something looked familiar in your video, it is my License code.  I traced it down to a address part of the DFU.  Knowing this, I have redone the bin files with your License code.  Attached are: 
* SergioDS203_P-D346CS164A113stock.bin - full image latest factory fw
* Sergio_HW_V282_DFU_V346C_WC_V65.BIN - full image Wildcat 6.5 fw https://github.com/MotoMaxis/DS203-DSOQuad
* SergioDS203.DFU.V346C.bin - only DFU 3.46C

With these files you should not get the license error.
[SergioDS203.zip](https://github.com/gabonator/LA104/files/5727581/SergioDS203.zip)

Sergio-, 2020-12-22:
Hi MotoMaxis. Thank you very much about all the effort on solving this. Really appreciate it.
I'm happy to report that flashing Sergio_HW_V282_DFU_V346C_WC_V65.BIN solved most of the issues. Not license or FPGA errors. APP seems to be working OK.
Except for a little thing: It seems I"m not able to save a configuration file. When I try the "Save File"000.CFG it flashes ERR. That's it.
I"m not sure if there's anything I can do to format the DFU
I also noticed that, when mounted, the serial number is kind of scrambled.
The Disk size is ~2Mb (instead of the Mb). I'm wondering if there's a way to force format it somehow.
[image]

Apart from that, all seems fine.
Thanks again for all the help!
PS: If you can just give some hints about the tools/utilities used to do the conversion, find the serial, etc, etc. I believe it will be beneficial as a learning resource.

Thanks a lot!

MotoMaxis, 2020-12-22:
Great news!

I think you can use gparted or fdisk or built in tool. https://vitux.com/how-to-format-a-usb-drive-in-ubuntu/

Try to remove 2mb partition first then recreate it. Then format.

Sergio-, 2020-12-22:
@MotoMaxis When you mount the DFU disk, does it show up as 2 or 8Mb?
Thank you.

Sergio-, 2020-12-22:
I tried with different partition types (FAT12, FAT16, FAT32, NTFS, etc).

Although I can read and write data using the drive mounted on the PC, in none of those the Option of "Save File" worked on the application. 

Here's the details of the current format I stick with. 

Any suggestion on how to fix the save configuration file is appreciated!

Thank you.

```
focal@ES1:~$ sudo fdisk /dev/sdb

Welcome to fdisk (util-linux 2.34).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): i
Selected partition 1
         Device: /dev/sdb1
          Start: 256
            End: 2047
        Sectors: 1792
      Cylinders: 113
           Size: 7M
             Id: 1
           Type: FAT12
    Start-C/H/S: 16/0/1
      End-C/H/S: 127/0/16

Command (m for help): 

```

MotoMaxis, 2020-12-22:
After formating, is the disks serial number readable now?

This applies to Wildcat 6.5 firmware...
After format and restart, I get the Configuration file not found error. I do the option Save 000.cfg. This will create serialnumber.wpt file. This is the default config.  When I tried saving to 001.cfg, I get ERR. You need to copy default .wpt to 001.cfg manually, then you can save to it and load it. I think this is unfinished part of firmware. Maybe someone can fix it on https://github.com/MotoMaxis/DS203-DSOQuad

The other Save BUF, DAT, CVS, ROM show OK. You have to remount to see new files.

Here is how my 8MB storage disk look.
`
DISKPART> list volume

  Volume ###  Ltr  Label        Fs     Type        Size     Status     Info
  ----------  ---  -----------  -----  ----------  -------  ---------  --------
  **Volume 5     I   DS203        FAT    Removable      8 MB  Healthy**

DISKPART> FILESYSTEMS

Current File System

  Type                 : FAT
  Allocation Unit Size : 4096
  Flags : 00000000

I:\>dir
 Volume in drive I is DS203
 Volume Serial Number is DC3B-1D3B
`

Sergio-, 2020-12-22:
Hi MotoMaxis
So close, yet so far. :-)
The disk seems to be working ok on the PC, I can save store/retrieve files, read and write it without issues.

[image]

The Problem seems to be something on the firmware not able to access the filesystem correctly, I tried the Button 3 method. It doesn't save anything on the Disk (always shows ERR). I also tried saving CSV, ROM and some other format, always ERR, umounting/mounting disk again always show it as empty.

On the Linux log, I still see the serial number like this: 

[image]

This is the result of fdisk -l:
```
focal@ES1:/media/focal/28F4-0161$ sudo fdisk -l /dev/sdc
Disk /dev/sdc: 8 MiB, 8388608 bytes, 2048 sectors
Disk model: Flash Disk      
Units: sectors of 1 * 4096 = 4096 bytes
Sector size (logical/physical): 4096 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: dos
Disk identifier: 0xaa68d4b6

Device     Boot Start   End Sectors Size Id Type
/dev/sdc1         256  2047    1792   7M  1 FAT12
focal@ES1:/media/focal/28F4-0161$ 

```

All my investigation points to some disconnect between the firmware and the filesystem, but I can't comment much further as I'm not familiar with the ST's and the firmware architecture. 

I"m ready to try whatever you guys think it's a good idea.

Thank you.

MotoMaxis, 2020-12-23:
I suggest you try the SergioDS203_P-D346CS164A113P10F26A(27AB).bin image.  That is factory code.  See if Saving files works. Try to delete 8MB partition and create a new one.  I am wondering if the 8MB flash memory is corrupt. Or if its info somewhere inside the STM.

The Wildcat firmware checks for a string 8MB Internal in order to decide to use 4096/sec for 8MB or 512/sec for 2MB.  Maybe it defaults to 512 for you.

Sergio-, 2020-12-23:
Hi MotoMaxis. I did as you suggested. I flashed the SergioDS203_P-D346CS164A113P10F26A(27AB).bin and flashed the FPGA, SYS, APP, etc, all using the st32flash. All ok. Removed the partition and create it again. Rebooted, but unfortunately, the firmware still not able to create the files on the disk (when hitting the Save Param for example, it shows up the message "Load File Fail").

MotoMaxis, 2020-12-23:
I am thinking perhaps linux is formating the flash in a different way. I can dump the flash memory into a bin and if you know a way to write it to yours, we can try that. Or you should find a Windows machine and format it there.

Quas7, 2020-12-23:
not sure, if this is worth anything but seems related and maybe it gets you a tiny bit closer: #26 
You would have do adapt the command a bit for the 8MB, I guess.

gabonator, 2020-12-23:
I am using this command to format LA104 fs, and the DS203 8MB has the same sector size, so maybe it will help:

sudo newfs_msdos -F 12 -v la104 -O "MSDOS5.0" -S 4096 /dev/disk2

-F 12 (fat12)
-v la104 (volume label)
-O "MSDOS5.0" (oem name, otherwise it will use BSD, but the device does not care anyway)
-S 4096 (sector size)

Sergio-, 2020-12-23:
Hi Folks. Thanks for the suggestions.

The command below did the trick on my Linux machine:
`sudo mkfs.msdos -s 1 -n DSO203 -f 2 -v -F 12 -S 4096 -r 512 -I /dev/sdb`

I believe with that all the functions are back to normal. I didn't start using the device, but I believe it should be fine.

Thanks a lot @MotoMaxis @gabonator @Quas7 for the effort in helping restoring the device. I believe this thread now contains a good knowledge to help others with similar issues. :-)

@gabonator Back to the original theme of this thread... Is there a way to install your version on the v2.82 Hardware? Is there any information I can extract from my device that can help you with that?

Thanks a lot.

MotoMaxis, 2020-12-23:
The DFU V3.46C stores the License code of a device at 0xFC to 0xFF in reverse (Little Endian) format.  You can use a hex editor to enter your own code.  
Sergio's example DFU = 3.46C ser: 8726B9A3 lic: 7A9457A7
[image]

MotoMaxis, 2020-12-24:
Hi @Sergio and anyone else, could you test this Wildcat V6.5 modified firmware on a 2.81 or 2.82 device?

**Make sure there are no CONF00x.CFG files on 8MB drive.**  We are testing new file creation. 
**Them try to Save CFG** (press&hold button 3) to a 001 to 009 (left wheel) and press button 3, do you get OK now?

[WC6.5testapp1.zip](https://github.com/gabonator/LA104/files/5738492/WC6.5testapp1.zip)

Sergio-, 2020-12-24:
Hi @MotoMaxis Yes. It worked.
I also must say that, the WildCat 6.5 you had prepared for me before, also worked after I formatted the disk with mkfs.msdos.
Have a nice day!

MotoMaxis, 2020-12-24:
Actually the before version only works for default 000.cfg. But fails for others unless conf001.cfg to 009 already exist. The test version has a minor update to automatically create new files for 001 to 009.

[21 more replies]

closed 2021-02-08 by gabonator