gabonator/LA104 #29: LA104 impossible to compile

nieldk, 2020-12-09:
While the release files flash and installs fine, it seems more files are missing in the repo.

For example, 17analyser,which should be the default LA104 analyser app is missing. 

So, compilation wont finish. 

Iam using the build script supplies in tools which should create hex file and zjp file with root filsystem.

Am i missing something

gabonator, 2020-12-09:
See the readme, the first application is the analyser app you are referring to:
[image]

And it is placed here:
https://github.com/gabonator/LA104/tree/master/system/apps/test17_official

To build the OS and all apps, run this script:
https://github.com/gabonator/LA104/blob/master/system/scripts/build.sh

And if the build fails, paste here the console output please

nieldk, 2020-12-09:
Yes, build.sh calls https://github.com/gabonator/LA104/blob/master/system/scripts/release_image_la104.sh which fails first in line 38 - "cp $FROM/17analys.elf $TO/analyser.elf" 
17analys.elf doesnt exist.
There are a couple more like that, so, in order to compile i will need to locate the correct filenanes?

gabonator, 2020-12-09:
build.sh builds the os and all the apps and copies the binaries into one folder (build/apps) for the packager script. Check the contents of the “build” folder and paste here the script output, it should point us what went wrong

nieldk, 2020-12-10:
For the record, I am trying on Ubuntu 20.10, using gcc-arm-none-eabi-9-2020-q2 

These are the fails from running https://github.com/gabonator/LA104/blob/master/system/scripts/build.sh

```[WARNING] ../apps_featured/103_servo/: Build failed
[WARNING] ../apps_featured/80_rftool/: Build failed
[WARNING] ../apps_experiments/102_ssd1306/: Build failed
[WARNING] ../apps_experiments/93_mfrc522/: Build failed
[WARNING] ../apps/73_invt/: Build failed
[WARNING] ../apps/79_cc1101osc/: Build failed
[WARNING] ../apps/test17_official/: Build failed
```

Output from each of these builds, running the build.sh in each application.

```nieldk@x201:~/build/LA104/system/apps_featured/103_servo$ ./build.sh 
../source/main.cpp: In member function 'virtual CRect CDialConfig::GetSubitemRect(int)':
../source/main.cpp:206:5: error: control reaches end of non-void function [-Werror=return-type]
  206 |     }
      |     ^
In file included from ../../../os_library/include/../../os_host/source/bios/../framework/Classes.h:2,
                 from ../../../os_library/include/../../os_host/source/bios/Bios.h:2,
                 from ../../../os_library/include/library.h:2,
                 from ../source/main.cpp:1:
../source/main.cpp: In member function 'int CItemFrameInfo::EasingFunction(int)':
../../../os_library/include/../../os_host/source/bios/../framework/Types.h:19:46: error: control reaches end of non-void function [-Werror=return-type]
   19 | #define _ASSERT(e) {if(!(e)) _HandleAssertion(__FILE__, __LINE__, _STR(e)); }
      |                              ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../source/main.cpp:552:17: note: in expansion of macro '_ASSERT'
  552 |                 _ASSERT(0);
      |                 ^~~~~~~
../source/main.cpp: In member function 'const char* CDataPoint::GetEasingName()':
../source/main.cpp:126:5: error: control reaches end of non-void function [-Werror=return-type]
  126 |     }
      |     ^
cc1plus: all warnings being treated as errors`

nieldk@x201:~/build/LA104/system/apps_featured/80_rftool$ ./build.sh 
In file included from ../source/main.cpp:17:
../source/modem/cc1101.h: In member function 'virtual bool CSampler::Init()':
../source/modem/cc1101.h:7:24: error: no return statement in function returning non-void [-Werror=return-type]
    7 |   virtual bool Init() {}
      |                        ^
../source/modem/cc1101.h: In member function 'virtual bool CSampler::Read()':
../source/modem/cc1101.h:9:24: error: no return statement in function returning non-void [-Werror=return-type]
    9 |   virtual bool Read() {}
      |                        ^
../source/main.cpp: In function 'bool analyse(CArray<short unsigned int>&)':
../source/main.cpp:465:1: error: no return statement in function returning non-void [-Werror=return-type]
  465 | }
      | ^
cc1plus: all warnings being treated as errors
arm-none-eabi-gcc: error: ./main.o: No such file or directory
arm-none-eabi-objdump: 'output.elf': No such file
readelf: Error: 'output.elf': No such file
./build_la104.sh: line 28: 123887 Segmentation fault      (core dumped) ../../../../tools/elfstrip/elfstrip output.elf 80rftool.elf
nm: 'output.elf': No such file
nm: 'output.elf': No such file
nm: 'output.elf': No such file`

nieldk@x201:~/build/LA104/system/apps_experiments/102_ssd1306$ ./build.sh 
../source/Adafruit_SSD1306.cpp: In member function 'uint8_t Adafruit_SSD1306::setBuffer(uint8_t*)':
../source/Adafruit_SSD1306.cpp:605:65: error: no return statement in function returning non-void [-Werror=return-type]
  605 | uint8_t Adafruit_SSD1306::setBuffer(uint8_t* buf) {buffer = buf;}
      |                                                                 ^
cc1plus: all warnings being treated as errors

nieldk@x201:~/build/LA104/system/apps_experiments/93_mfrc522$ ./build.sh 
../source/main.cpp: In member function 'bool CWire::error()':
../source/main.cpp:27:1: error: no return statement in function returning non-void [-Werror=return-type]
[... 39 lines cut ...]
arm-none-eabi-objdump: 'output.elf': No such file
readelf: Error: 'output.elf': No such file
./build_la104.sh: line 32: 123982 Segmentation fault      (core dumped) ../../../../tools/elfstrip/elfstrip output.elf 79ccosc.elf
nm: 'output.elf': No such file
nm: 'output.elf': No such file
nm: 'output.elf': No such file

nieldk@x201:~/build/LA104/system/apps/test17_official$ ./build.sh 
../source/Analyze.c: In function 'u16 I2C_AnalyzeStartCnt(u64)':
../source/Analyze.c:327:44: error: 'memset' used with length equal to number of elements without multiplication by element size [-Werror=memset-elt-size]
  327 |     memset(AnalyzeTime, 0, RECORD_DEPTH / 2);
      |                                            ^
../source/Analyze.c: In function 'u16 SPI_AnalyzeDataStartCnt(u64)':
../source/Analyze.c:692:44: error: 'memset' used with length equal to number of elements without multiplication by element size [-Werror=memset-elt-size]
  692 |     memset(AnalyzeTime, 0, RECORD_DEPTH / 2);
      |                                            ^
cc1plus: all warnings being treated as errors
arm-none-eabi-gcc: error: ./Analyze.o: No such file or directory
arm-none-eabi-objdump: 'output.elf': No such file
./build.sh: line 18: 124020 Segmentation fault      (core dumped) ../../../../tools/elfstrip/elfstrip output.elf 17analys.elf
nm: 'output.elf': No such file
nm: 'output.elf': No such file
nm: 'output.elf': No such file
rm: cannot remove 'output.elf': No such file or directory`
```

nieldk, 2020-12-10:
Oh, my bad. Just noticed.
I should be using gcc-arm-none-eabi-7-2018-q2-update.

Well, who knows, maybe I will update to be able to use a more recent gcc ;)

gabonator, 2020-12-10:
Wow, interesting. I will check it later today. Of course we want to use latest GCC, and for sure we need to fix these issues. So I will reopen this issue until me or you or anyone will resolve these errors and achieve compatibility with the latest gcc. Thank you for reporting this

nieldk, 2020-12-10:
Closed with PR #30

closed 2020-12-10 by nieldk