gabonator/LA104 #36: UARTmonitor sends data

ArthurVnL, 2021-01-12:
Hi,

I am using the last LA104 stable release. The UARTmonitor is constantly sending out data like "Ahoj, toto je riadok 'counter' ", I think this was meant for the promo video. However I dont think it is supposed to be in the release, am I right here?

Thanks for the great repo!

gabonator, 2021-01-13:
Yes, I know about it. I designed the application in a way, that you can send prepared messages from a text file (e.g. common AT commands). But I did not finish this feature, so just for testing I left there just this message. At least for now I will translate it to English

gabonator, 2021-02-20:
Fixed, and added logging to file, see screenshots:
https://github.com/gabonator/LA104/tree/master/system/apps_featured/72_uartmon

closed 2021-02-20 by gabonator