gabonator/LA104 #37: Unable to copy the HEX file

RChadwick7, 2021-01-13:
When I copy the factory firmware HEX file, it copies over fine, and the hex file changes to *.RDY. However, with this firmware, The copy seems to start, hangs, and eventually errors out oddly, saying not enough space (Even though this firmware is smaller than factory). Then, the LA104 disconnects, and I get a USB error when it reconnects. I have Teracopy installed, and have disabled it. When I copy using Teracopy, it copies fine, but when the drive reconnects, the firmware file is changed to *.err. Both of these does seem to get the firmware on the device, as it somewhat works, but has quick crashes, and I'm guessing it's not fully installed.

BG7UHS, 2021-02-17:
We need to copy two files (.hex file and any blank text file). After copying, the extension of both files will become .err files.  At this time, delete the blank text file and change the firmware .err extension to .hex.  We will get the .rdy file.

gabonator, 2021-02-24:
Since there are more users having these problems, I made a discussion here: https://github.com/gabonator/LA104/discussions/45 

If you have found reliable method how to upload fw please share your thoughts there

closed 2021-02-24 by gabonator