gabonator/LA104 #38: "Asssertion Failed" errors when using some functions in the minigen and miniosc apps on the dso213

Steppan5g, 2021-02-04:
Has anyone had this issue before? I'm running the alternative firmware and some functions in minigen and miniosc apps don't do anything besides flashing briefly an "Assertion Failed" error. For example, when I try to change the waveform in the minigen I get "Assertion failed in ../source/../../../os_source/gui/Controls.h [267]:0". The same exact error shows up when I push K1 to show the side menus in the miniosc, or when I try to change settings in the side menus of the disp tab.

Thanks to whoever helps me

gabonator, 2021-02-04:
These applications are just ported from the DS203 firmware and not all functions are fully working. Assertion failed message means that something went wrong or something is not implemented yet. Normally the application could continue running, but I decided to freeze the app in that point so I will be forced to fix it properly.

Since I do not know if someone uses these apps, I focus on those which I find more valuable - mostly on new LA104 applications. When I will return to the development of DS203/DS213 I will fix it (but that will be probably not sooner than in 2 weeks), if you have basic C++ programming skills you can try fixing it on your own. Tutorials on how to setup building environment is in the readme

If you will find more problems, feel free to report here new issues.

Steppan5g, 2021-02-04:
Thanks! I'll update you if I find something else

robhubi, 2021-02-07:
The error that Steppan5g described also occurs with me.

Hardware
==============
DFU version: V3.73C
Hardware version: SCH_V2.0
System version: DS213
FPGA version: AGL1K
Display: ILI9327 (400x240)
Disk: 8MB (8188kB)

Software
==============
BIOS version: 0104
Build rev: a826b7c
Build date: Dec 23 2020 19:08:18
Build user: gabrielvalky
Build system: darwin19

nieldk, 2021-03-17:
I have seen that error on random occasions.

I started formatting the filesystem, every time I update my LA104, this seems to help.

Not sure if it's the same for the oscillator devices, but I assume it could be.

gabonator, 2021-03-17:
Please report here the assertion message and steps for reproducing

nieldk, 2021-03-20:
Well, latest was with RF analyser synthesizer, when trying to transmit a capture, after having modified one field (temperature)  Assertion failed in ../source/modem/cc1101.h [403]: len > 0 & len < 256

gabonator, 2021-03-20:
@nieldk please export captured signals with button "Save" in capture tab and paste here produced file

nieldk, 2021-03-20:
> @nieldk please export captured signals with button "Save" in capture tab and paste here produced file

`[{_protocol:"Nexus Temperature & Hum","length":36,"data_0":0x468029f2,"data_1":0x3000000,"id":70,"battery_low":1,"channel":1,"temperature10":41,"humidity":32,"_timestamp":555777,"_uid":1000},{_protocol:"TFA-Twin-Plus-30.3049","length":36,"data_0":0x5e0c2099,"data_1":0x30000000,"temperature10":44,"humidity":-3,"id":37,"battery_low":0,"channel":3,"valid":0,"_timestamp":742432,"_uid":1001,"junk":0}]`

gabonator, 2021-03-20:
@nieldk I confirm that it crashes when trying synthesizing nexus packet. Code responsible for synthesizing is simply commented out: https://github.com/gabonator/LA104/blob/master/system/apps_featured/80_rftool/source/protocol/nexus.h#L114

Unfortunately I am currently very busy to fix it. But I can guide you how to fix it and you can do it by yourself.

nieldk, 2021-03-21:
> @nieldk I confirm that it crashes when trying synthesizing nexus packet. Code responsible for synthesizing is simply commented out: [...]

Thanks, I might give it a look. But it's not major thing.

Alexxdal, 2021-07-21:
I fixed the minigen error in my fork https://github.com/Alexxdal/LA104

gabonator, 2021-07-22:
@Alexxdal feel free to open pull request with your changes