gabonator/LA104 #41: Record and view any wireless protocol (Feature Request)

RChadwick7, 2021-02-09:
The RF potential for this looks fantastic, but after adding a CC1101 board, I realized I don't have any compatible devices that are automatically recognized. It would be great if unrecognized devices could still be recorded, analyzed, and played back.

gabonator, 2021-02-10:
I was considering support for uknown protocols, but there are more reasons why this would be too challenging:
- The main feature for the rf toolkit is analyse and synthesize - you can receive a packet, modify it and send it out. There is no similar device or pc application that does this. This requires the app to "understand" the received signal 
- Implementing new protocol is quite easy, so the community could contribute to this project and add support for new devices, here is code for [FA-Twin-Plus weather station](https://github.com/gabonator/LA104/blob/master/system/apps_featured/80_rftool/source/protocol/weather.h)
- The device has very limited resources - storing arbitrary packets would consume too much RAM. Currently we only store few numbers needed to reconstruct the packet (type of device + temperature + humidity + device id...)
- There are two ways how to transmit data with CC1101 - either you use one of the input pins as OOK modulation input, or you transfer the paket through SPI interface. I decided to use SPI interface, because the application doesnt need to handle precise timing, you just let the transceiver to transmit the binary stream at specified speed from byte buffer provided through SPI
- For transmitting packets over SPI we need to know what is the shortest pulse duration, or we need to know what is the greatest common divisor of all possible pulse lengths to be able to construct the bitstream. And this is not possible with arbitrary protocols, we need to simply quantize the durations somehow

Of course we can add some universal analyser for manchester coding, gap coding and other common modulation types. There is one application that helps you identifying the protocol type [WaveAlchemy](https://github.com/gabonator/LA104#wave-alchemy), but it is very experimental and since we have CC1101 webusb rf analyser application I stopped the development of WaveAlchemy, because web browser offers much more options for analysis of unknown protocols

To analyse new protocols I use this workflow:
- Use spectrum analyser with directional antenna to identify new signal source and its precise frequency (not necessary, you can guess the frequency until you get stable signal)
- Use computer and Webusb rf analyser to catch few packets tuned at this frequency. This application integrates RTL433 library which has support for more than 400 protocols and if it will recoginze your signal, your are lucky
- Add new protocol to the Rf toolkit application based on the RTL433 decoder
- Check the decoder with packets from webusb rf analyser using unit test (https://github.com/gabonator/LA104/blob/master/system/apps_featured/80_rftool/test/test.cpp) - it tries to decode array of microsecond intervals and then encode it back and decode again to verify if decoder and encoder is consistent

For recording arbitrary packets I would rather make separate application (or continue developing WaveAlchemy app), but I am afraid that we cannot use the CC1101 in direct modulation mode using its GPIO pin, because there is no extra GPIO pin on LA104.

So for now we can keep this issue open until someone (or I) adds support for generic modulations.

cypresstwist, 2021-02-12:
I have a bunch of 868MHz LoRa devices and a LoRaWAN gateway on the same ISM band. How can I help? Could those be imported?

BG7UHS, 2021-02-17:
RF TOOL cannot capture PT2262 EV1527 chip data. Hope that more RF functions can be added.  For example, some functions of im-me.  [opensesame](https://github.com/samyk/opensesame)

gabonator, 2021-02-17:
Lora uses proprietary protocol with special modulation. This app works only for simple OOK modulations.

BG7UHS, 2021-02-17:
> RF TOOL无法捕获PT2262 EV1527芯片数据。希望可以添加更多的RF功能。例如，im-me的某些功能。 [芝麻开门](https://github.com/samyk/opensesame)

@gabonator

gabonator, 2021-02-17:
@BG7UHS as I noted above, everything is open sourced and with a little effort and light C++ skills it is possible to implement new decoders. Since I do not own any EV1527 transmitters, I cannot help with this.

art9tem, 2021-04-04:
> @BG7UHS Hello, please send me a message with the help of "wave alchemy" I want to conduct tests !

https://github.com/gabonator/LA104/blob/master/resources/waveAlchemy1.jpeg

BG7UHS, 2021-04-04:
> > @ BG7UHS您好，请在我要进行测试的“波浪炼金术”的帮助下给我发送消息！
> 
> https://github.com/gabonator/LA104/blob/master/resources/waveAlchemy1.jpeg

I don't quite understand what you mean.  Do you want me to test it for you?

art9tem, 2021-04-04:
> > > @ BG7UHS您好，请在我要进行测试的“波浪炼金术”的帮助下给我发送消息！
> > 
> > 
> > https://github.com/gabonator/LA104/blob/master/resources/waveAlchemy1.jpeg
> 
> I don't quite understand what you mean. Do you want me to test it for you?

I can't find the same functionality as in the screenshot . You can send me this app. Your article says it's in development.

BG7UHS, 2021-04-04:
Oh, this is not my article, the function you mentioned hasn't been implemented yet.  Still testing?  I can't find this feature either.  You have to ask the author. @gabonator

gabonator, 2021-04-04:
I did not include wave alchemy in releases, because it was only experimental project. It does not offer much functionality, it just shows histogram of the received signal to easy identify of modulation type. But if you want, you can build it by yourself: https://github.com/gabonator/LA104/tree/master/system/apps/test41_alchemy

art9tem, 2021-04-04:
> I did not include wave alchemy in releases, because it was only experimental project. It does not offer much functionality, it just shows histogram of the received signal to easy identify of [...]

How can I add this to my device in the menu ?

gabonator, 2021-04-04:
The easiest way is to make new folder in the root, copy new files there and run file manager. If you want to integrate it with gui, create "index.lnk" in that folder with following contents:

    {"description":{"short":"My folder","long":"My folder"},"icon":"folder.bmp",order:500}

where folder.bmp is the icon you want to use, and the higher is the order number, the sooner the icon appears on the screen (negative values put them at the end of the list). And then you create app.lnk file with similar contents:

    {"description":{short:"CiPython",long:"Circuit python"},"icon":"cpython.bmp","execute":"cpython.elf"}

RChadwick7, 2021-04-04:
> I did not include wave alchemy in releases, because it was only experimental project. It does not offer much functionality, it just shows histogram of the received signal to easy identify of [...]

Any chance you might decide to include it in future releases?

gabonator, 2021-04-04:
@RChadwick7 the purpose of wave alchemy was to have universal tool for reception, analysis and synthesis (replay attack) of radio or infrared signals. Yes, the nature of the TV remote signal is the same as OOK modulation, it just uses different media for transmission. The histogram view is great way how to visualize the pulse durations and you can immediatelly distinguish between various kinds of codes (gap coding/manchester coding/...) and this helps you to develop decoding algorithm for unknown modulations/codings.
I needed more powerful tool which could automatically decode many known algorithms (rtl433 library), but it wasn't possible to integrate it into LA104 with very limited memory size. After developing webusb applications for DS203/DS213 oscilloscopes I realized this is the way how to finish the project - all the hard work will be performed on computer (in javascript) instead of LA104. The LA104 will only sample the output signal of CC1101 demodulator at fixed speed and the web application can automatically decode identified signals using available decoding libraries. After porting [rtl433 decoder](https://github.com/merbanan/rtl_433) to web assembly, I integrated it with the webusb application to form very powerful signal decoding tool. 
So I currently do not plan to continue developing the wave alchemy, but I encorauge everyone with even little coding skills to continue improving this application

art9tem, 2021-04-07:
help me I can't connect the la104 to the web interface to decrypt the signal

gabonator, 2021-04-07:
@art9tem the usb cable does not fit?

art9tem, 2021-04-08:
> @art9tem the usb cable does not fit?

Would you like to try to buy another antenna with a frequency of 125 kHz and make a replay attack with Keyless-Go. I think it's a good idea