gabonator/LA104 #43: [Feature Request] Upload firmware to target with SWD

203Null, 2021-02-21:
Not sure if this is doable but if it works this is gonna be very handy.

gabonator, 2021-02-22:
It is, I tried it on DS213, but it should work for LA104 as well. Both devices have SWD interface attached to USB port connector, see this:

[image]

full schematic:
https://github.com/gabonator/LA104/blob/master/resources/LA104%20V1.5B_SCH.pdf

use openocd to upload fw over SWD innterface:
https://github.com/gabonator/LA104/tree/master/tools/openocd

203Null, 2021-02-22:
I mean upload to another STM32 target using LA104's GPIO. Not upload to the LA104/DS213. Sorry for the ambiguous wording.

203Null, 2021-02-22:
I don't know that it's OK to use short USB with SWD. Won't this block SWD debugging while USB is in use?

gabonator, 2021-02-23:
I was surprised as well to see SWD tied to USB data lines, but it is working well. Regarding the stlink v2 application on LA104 - it is definitelly doable - it is just using two digital lines, but I could not find any source code of the programmer.

Turro75, 2021-03-06:
In the past I used the black magic probe (https://github.com/blacksphere/blackmagic under GPL3) loaded on a stlink v2.
This is also suitable for samd21 (i.e. arduino zero)