gabonator/LA104 #50: DS203 2020.11.4 release shell.elf Assertion failed

erik-smit, 2021-03-11:
I have a DSO Quad DS203 V2.7

When starting shell.elf I get the error: `Symbol '_ZN4BIOS3SYS12GetAttributeENS0_10EAttributE' not found!`.

I suspect this is because my install with dfuload fails:
```
$ ./dfuload /dev/sdb cp ~/Downloads/system_ds203.hex 
Opening: '/dev/sdb' .....................................................
Failed to open disk '/dev/sdb', errno:13
```

I've tried flashing STM32 via UART, but have not been succesfull.
Copy .hex with W10 also fails. 
Copy .hex with mtools gives no error:

```
[smite@ErikXPS13 dfuload]$ sudo mcopy -i /dev/sdb ~/Downloads/system_ds203.hex ::system_ds203.hex
[smite@ErikXPS13 dfuload]$ sudo mdir -i /dev/sdb
 Volume in drive : is DFU V3_11_C
Directory for ::/

SYSTEM~1 RDY     74413 2021-03-11  15:21 
        1 file               74 413 bytes
```
But still same error.

My DFU is the famous V3.11C. Is this "symbol not found" caused by install not fully completing?
Should I contain to try to flash my STM32 to fix this problem?

gabonator, 2021-03-11:
symbol not found means you have different version of system (firmware hex file) and application (shell.elf). Download newest package from releases. regarding the problems with fw upload - try to use the dfuload tool. Error 13 means permission denied, so try sudo dfuload...

erik-smit, 2021-03-11:
Ohhh, the progressbar as dots with dfuload made me think it was writing, but it was not.
It's ok now, thank you!

closed 2021-03-11 by erik-smit