gabonator/LA104 #51: DS203 WebUSB "Interface handle not available for control transfer"

erik-smit, 2021-03-12:
Trying to use the WebUSB Oscilloscope on my DS203 (HW v2.7b) on Windows 10 with Chrome.

When I plug it in, I get the Chrome popup "DS203 Oscilloscope detected", but the resulting page fails to connect to the scope.
In developer tools, I see: "Uncaught (in promise) DOMException: A transfer error has occurred.".

Because the popup appears, I think the Zadig WinUSB installation was fine.

In "chrome://device-log/" I read:
```USBError[20:47:12] Interface handle not available for control transfer.
USBUser[20:46:27] USB device added: path=\\?\usb#vid_1209&pid_db42#197424433258353037ff717#{a5dcbf10-6530-11d2-901f-00c04fb951ed} vendor=4617 "Gabriel", product=56130 "DS203 Oscilloscope", serial="197424433258353037FF717", driver="WinUSB", guid=e578864b-94a7-454d-b10e-ef3d10a64703```

This might but the problem, but I don't really know where to begin debugging USB. 
Also, this particular error only have hits in the Chromium source on Google: https://www.google.com/search?q=%22Interface+handle+not+available+for+control+transfer.%22

In "chrome://usb-internals/" I don't see any problems inspecting the DS203 device.

And idea how to continue debugging or solving this?

I've tried both the last 'official' DS203 release and building master with the docker.

erik-smit, 2021-03-12:
Ahh, after some fiddling with udev, I've been able to get it working on Linux.
So, yeah, it works on Linux. Woo!
But W10, not yet.

gabonator, 2021-03-13:
Tested today on Win7 machine (DS213) and I confirm that it just shows the same error message in output log. Bit strange for me since I remember it worked well when I tested it in early stage of development of the application with zadig driver

gabonator, 2021-03-13:
The problem is in the function controlTransferOut (serial.js:68). Just tried to omit calling this function and everything works well. For a simple workaround one can set break point in developer tools in serial.js on line 39, where is line "return this.device_.open()" and then type in the console this command and let the program continue (Resume script execution F8)
`this.device_.controlTransferOut = () => Promise.resolve()`
I need to do more investigation to find out what could be the universal solution for both platforms.

gabonator, 2021-03-13:
Fixed in commit https://github.com/gabonator/LA104/commit/5c168b5f01b626740e99faeb148872dbff72d2b4 No update is necessary
@erik-smit could you please verify it on your windows machine?

erik-smit, 2021-03-13:
> @erik-smit could you please verify it on your windows machine?
The error is gone and I'm getting output on Windows!

closed 2021-03-13 by erik-smit