gabonator/LA104 #54: Convert miniware .hex to compatible format?

a7hybnj2, 2021-03-26:
Is there a way to take the .hex files provided by miniware and allow them to run from shell.elf?

I know they source they have released is several version behind the .hex files they have made available.

Thanks!

Kinda related to #89 but I also want to run the more recent version of the LA104 analyzer also (with working screen capture hotkeys). Also, these mods would be pretty sweet. https://github.com/Ralim/LA104

THANKS!

gabonator, 2021-03-26:
Yes there is. You need to take the source code and rebuild it. It was already done here (with original fw):

https://github.com/gabonator/LA104/tree/master/system/apps/test17_official/source

But since the filesystem functions work differently in this OS, I removed all import/export functions. It it not difficult to fix it, but not planned from my side. I would rather spend time developing new signal analyser app with more intuitive UI, but I do not plan to do it in near future (too busy with other stuff)

a7hybnj2, 2021-03-26:
@gabonator wow thanks for the fast reply! This is such an awesome project!

I guess what I was trying to ask was: Since, miniware hasn't release the source code for the **latest** firmwares...if there was a way to take their compiled .hex files and have them load from your OS.

Or maybe boot while holding F2 would get you into default firmware and regular boot would be shell.elf... Anyway I am just throwing around ideas. I want to switch to the alternate DS213 OS but I know they fixed some things in the newest firmware so I am hesitant.

gabonator, 2021-03-26:
I see. But there is no reason not to publish the source code. So you can try asking for it. Unfortunately there is no easy solution how to load HEX firmware. Because every application loads to a specific range of FLASH memory and uses specific range of RAM. And I am sure that these ranges would be in conflict with the location where the OS is loaded. So the only safe way how to run it is rebuild it with the available memory regions in mind. holding F2 could do the trick, but again they need to provide the HEX file compiled for the second app slot

nieldk, 2021-03-27:
> I see. But there is no reason not to publish the source code. So you can try asking for it. Unfortunately there is no easy solution how to load HEX firmware. Because every application loads to a [...]

That part, holding F2 is interesting. It should be fairly simple to compile the source for this. Although, I can't recall seeing it done yet.
But, have a look at other FW, DS202 for example, this is implemented, and APP1/APP2 is as simple as changing a line in the source code.

a7hybnj2, 2021-03-27:
> But there is no reason not to publish the source code. So you can try asking for it.

Reading through their forums it looks like many people have asked over the last $timeframe and they still haven't. The source is version 1.22 and current .hex is 1.27 (for the DS213). So, not a ton of revision but they have appeared to fix the major bugs.

For the LA104 it looks like the source and latest .hex match @ 1.03 which is ahead by 2 minor revisions according to the "1.01" that pops up when I start 'logic analyzer' from my shell.elf.

> It should be fairly simple to compile the source for this.

Holding the button would be cool because I would choose to boot into the default firmware normally and "extra options" on hotkey-boot. ex: default oscope software on boot and webusbosc on hotkey would be amazin! But, I would still need the updated source code from miniware which they haven't provided yet. There are some other user versions of the ds213 firmware that look neat like the rolling update one and the one I linked above.

Anyway I don't really know what I am doing when it comes to this type of stuff but I might try getting compiling locally setup so I can start to poke around with it.