gabonator/LA104 #56: FR and Issue

mikerocklewitz, 2021-04-05:
Just wanna say this is absolutely phenomenal firmware.  I was on the fence about buying the LA104 and it put me over the edge.

When messing with the webusb rf analyzer, I can connect to my rf adapter, when I plug into usb I only see a com port. Nothing else happens with it.  Am I missing something obvious?

Feature request:
In the pwm page under gpio control, or a different page or app altogether - would it be possible to implement a function generator?

gabonator, 2021-04-05:
Please check this thread, there is a bug in javascript code which can be easily fixed: https://github.com/gabonator/LA104/issues/51

LA104 has only 4 universal input/output pins which can be configured to give either 0V or 3V, nothing in between. The only way how to generate voltage of specific value, which is necessary for a function generator, you need to use RC filter on PWM output or DAC converter. 

I am planning to make app for I2S DAC stereo converter which could be used as function generator in audible frequency range. But as a more universal solution one cold use DDS generator expansion board attached to LA104 (e.g. AD9833)

mikerocklewitz, 2021-04-05:
Thank you for the very informative response. Im not that far into the
device to know the ins and outs yet.

The ad9833 would work.