gabonator/LA104 #64: Lots of hardfaults on DS213

Red-M, 2021-06-28:
I've built the firmware from the current release and tried the latest available and I keep getting hardfaults for example when plugging in the USB for the webosc program.

Output from devinfo:
Hardware
==============
DFU version: V3.73C
Hardware version: SCH_V2.0
System version: DS213
FPGA version: AGL1K
Display: ILI9327 (400x240)
Disk: 8MB (8188kB)

Alexxdal, 2021-07-23:
the hardfault happen if you restart the device, open the webosc and then plug the usb in? 
check if it happen only when you first open the miniosc app, if yes wait for merge i opened a pull request that fix this errors

Red-M, 2021-07-24:
Yes, I tested exactly that.
I'll put your PR down and build it to test it.

Red-M, 2021-07-24:
I've tested your fork, it also produces hard faults.

Alexxdal, 2021-07-24:
can you post a screenshot of yhe hard fault?

Red-M, 2021-07-24:
[image]

Red-M, 2022-08-10:
I've recompiled the latest master as of right now and the oscilloscope app now works but the webosc doesn't which is a shame but I guess this is better than nothing.