gabonator/LA104 #65: Decoding SMBus

RChadwick7, 2021-07-10:
Can the new firmware decode SMBus to see laptop battery details?

gabonator, 2021-07-11:
No, I recommend using DS203 or DS213 with usb oscilloscope and writing custom protocol decoder in javascript.

RChadwick7, 2021-07-15:
Is it that the LA104 just doesn't have enough memory? It looks like SMBus is fancy I2C, Arduino project here: https://github.com/PowerCartel/PackProbe

Do you know of a cheap off-the-shelf logic analyzer that can do SMBus and others?

Quas7, 2021-07-15:
sigrok with a $8 LA should do it.
I did not find smbus on the list of sigrok decoders but I assume it is just hidden somewhere in the i2c decoder.

https://sigrok.org/wiki/Logic_analyzer_comparison

RChadwick7, 2021-07-16:
Thanks! I wasn't aware of Sigrok. Can you point me to the $8 LA?

Quas7, 2021-07-16:
there are clones of this one available.

https://sigrok.org/wiki/VKTECH_saleae_clone

I have one of those working fine besides some others LAs.

https://www.amazon.de/dp/B01MUFRHQ2/ref=cm_sw_r_cp_apa_glt_i_EH01SQ1NNEJM04JC58SY?_encoding=UTF8&psc=1

gabonator, 2021-07-18:
@RChadwick7 looking at the "PackProbe" code it looks quite easy to port it to LA104, but I do not own any machine with easily accessible SMBus, so I could not verify whether it is working or not