gabonator/LA104 #66: hex file crashes usb connection

bendelathouwer, 2021-07-16:
when I try ti copy the system_la104.hex to my la104. the usb connection times out after a while. does anybody know what is happening?

Kind regard,

Ben

closed 2021-07-17 by bendelathouwer

fenyvesi, 2021-08-23:
The same here. Windows10. Copying starts but hangs. After a while the device disconnects and unknown USB device appears that is not connected. After restarting LA104 the message is: FPGA OK, missing elf.
Made several times, the result is the same.
Windows Update?

gabonator, 2021-08-23:
Yes, the upload is somehow unstable. I had to copy the hex file multiple times on windows machine until it was working. It was always resulting in ERR file, but it was working. Don't know what is the reason - probably buggy implementation of the code provided by manufacturer. The only suggestion I can provide is to try it on another machine (with older version windows). Funny fact that it is impossible to flash it on Mac M1 based laptop - it is repeatedly connecting and disconnecting in quick intervals.

fenyvesi, 2021-08-23:
Thnaks for the rapid answer.
With the 2-files method I manged to get the system.rdy. But after a power-cycle the message is: FPGA Conf OK, missing shell.elf.
Not the missing elf is the problem?

gabonator, 2021-08-23:
That means OS is loaded and it cannot find "shell.elf" file in the internal flash disk. Just copy the contents of zip file from releases page to your device over USB - the OS takes over the USB and now it works like 8MB flash disk. In the root there should be "shell.elf" file which is the program that is automatically executed upon OS startup

fenyvesi, 2021-08-23:
OK, works after copying the content of the zip file.

Thanks for the sw and taking care of me.