gabonator/LA104 #77: LA104 WebUSB connection problem

fenyvesi, 2021-08-24:
WinUSB driver installed (Zadig v1.5). 
WEBUSB RF Analyser started. 
When asks in a window, then I connect to USB, and I can see the LA104 device in Windows Device Manager. 
The Chrome detects the unit and starts the web application that can't connect to the device. The sigin in the last line rotates endlessly.
Any idea? Did I do smething wrong?
Thanks.

gabonator, 2021-08-24:
What does it show in the debug console?

fenyvesi, 2021-08-25:
index.html:1 Uncaught (in promise) DOMException: A transfer error has occurred.

DevTools failed to load source map: Could not load content for wasm://wasm/rtl433.wasm.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME

fenyvesi, 2021-12-06:
Any development?

gabonator, 2021-12-06:
Unfortunately I was busy with other things... I did some tests initially, but without success and then I gave up... I will definitely have to look at it, but don't know when.

fenyvesi, 2021-12-07:
Thanks. Fenyvesi György Feladó: Gabriel ValkyElküldve: 2021. december 6., hétfő 22:26Címzett: gabonator/LA104Másolatot kap: George Fenyvesi; AuthorTárgy: Re: [gabonator/LA104] LA104 WebUSB connection problem (#77) Unfortunately I was busy with other things... I did some tests initially, but without success and then I gave up.. I will definitely have to look at it, but don't know when.

gabonator, 2021-12-11:
@fenyvesi could you please try this app:
[67webusb.elf.zip](https://github.com/gabonator/LA104/files/7696979/67webusb.elf.zip)

Use following web page to test the communication. Clicking on LED checkbox should show numbers '1' or '0' on screen.

fenyvesi, 2021-12-11:
Sure. I copied the elf file in the root directory. Started through file manager.
USB cable is connected.
On the top of the screen: USB begin............. a lot and more and more ...
On the bottom: Image loaded at 0x 080200000

What shall I do?

gabonator, 2021-12-11:
1. Run the app on device
2. Connect to computer (it must be disconnected before running the app)
3. Open the website from the link above
4. Click connect (it should display list of available devices, only one should be available to select)
5. Click on the LED checkbox and it should display '0' or '1' on the device screen
6. If anything won't work, paste here the console log from chrome

I do not have windows machine available right now, so I cannot verify it...
If there will be "Access denied" error in the console, try installing [Zadig driver](https://zadig.akeo.ie/)

fenyvesi, 2021-12-11:
OK we crossed. I am trying.

Steps 1,2,3 are done.
Step 4: "connecting" message disappears, nothing happens.

I try the zadig driver and coming back.

Chrome console:
Uncaught (in promise) DOMException: Access denied.
index.html:1 Uncaught (in promise) DOMException: The device must be opened first.

fenyvesi, 2021-12-11:
Replaced with WinUSB driver. The device name is strange: DABBoot DFU bootloader.
Can't connect, log:

index.html:1 Uncaught (in promise) DOMException: A transfer error has occurred.

gabonator, 2021-12-11:
"DAPBoot DFU Bootloader" is OK, this project borrowed from https://github.com/devanlai/dapboot and ported to LA104 OS. But the error message does not look good. Is it displayed every time you press the checkbox?

fenyvesi, 2021-12-11:
yes.

gabonator, 2021-12-11:
Try this one
[67webusb.elf.zip](https://github.com/gabonator/LA104/files/7697499/67webusb.elf.zip)

fenyvesi, 2021-12-11:
Practically the same.

Log each time I  press connect:
index.html:1 Uncaught (in promise) DOMException: A transfer error has occurred.

gabonator, 2021-12-11:
Ok, it seems that during porting the code to LA104 OS I introduced some bug there and the winusb component is not configured well. I don't know how to fix it, so I will try to port the latest version of dapboot tool again... As soon I will have something ready to test I will let you know. Probably tomorrow

gabonator, 2021-12-11:
I have an idea, try this:
https://rawgit.valky.eu/gabonator/LA104/master/system/apps_usb/test67_webusb/web/index2.html

fenyvesi, 2021-12-11:
When I start the last elf, connect USB, open the URL in chrome I can see:
Disconnect[]
The log is empty.

I push the Disconnect button and the log:

index2.html:1 Uncaught (in promise) DOMException: A transfer error has occurred

Does it mean it is connected?

gabonator, 2021-12-11:
Great! That means it is working!! So we were focused on wrong side of the problem - the application on LA104 is working, and the issue was in the web page

fenyvesi, 2021-12-11:
Yeah, the problem is not always there as we suggest. Hurray.

fenyvesi, 2021-12-11:
One remark. Should the URL arise automatically or I have to type each time?

gabonator, 2021-12-11:
But what am I confused now is, that whether it is necessary to do any changes in the LA104 app, please check this one:
[67-6wusb.elf.zip](https://github.com/gabonator/LA104/files/7697691/67-6wusb.elf.zip)

This application is just for testing, so you should refresh the website every time. The webpage in popup points to the original dap bootloader. In CC1101 webusb it should open the right page automatically

fenyvesi, 2021-12-12:
It works with the last URL. Your suggestion seems to be true.

gabonator, 2021-12-12:
AAaaah, hell. So it was working all the time, but I just forgot to add windows workaround into the web page javascript code. Pretty sad that we wasted time on this... I will push updated serial.js into the repo and consider closing this issue if it will work.

And thank you for helping with this issue

The RF webusb app is terrible, do not press other buttons than connect, init, start, stop. When editing frequencies just press TAB button, for the new value to be applied. When you have some clear signal in the scroll view, pause it with "Stop" button and highlight it with dragging your mouse. If it is some known protocol, it should display decode info in lower window. Open console window to see the precise intervals/timing of the packet

fenyvesi, 2021-12-12:
OK. Thanks, once more.

fenyvesi, 2021-12-12:
I can't really use it, unexpected serial errors occur.

gabonator, 2021-12-12:
Send screenshots, the rolling window with signal works? When the errors appear?

fenyvesi, 2021-12-12:
I downloaded the https://rawgit.valky.eu/gabonator/LA104/master/system/apps/79_cc1101osc/web/index.html page locally and use it locally with the modified serial2.js.
The problem as I see the rtl433.wasm is not downloaded and that is why is not found locally.

On the contrary in 1 case from 10 it captured and analysed signals but I can't repeat it.

The log:
Access to XMLHttpRequest at 'file:///C:/Users/x/Downloads/valky/rtl433.wasm' from origin 'null' has been blocked by CORS policy: Cross origin requests are only supported for protocol schemes: http, data, chrome, chrome-extension, chrome-untrusted, https.
decoder.js:18 GET file:///C:/Users/x/Downloads/valky/rtl433.wasm net::ERR_FAILED
(anonymous) @ decoder.js:18
readAsync @ decoder.js:5
Decoder @ decoder.js:71
(anonymous) @ analyser.js:243

gabonator, 2021-12-12:
Try the online version, I fixed the serial.js issue, flush the browser cache in case it will display the old version. If you want to do experiments locally, you need to use some sort of webserver and open it through localhost (I am using httpserver from nodejs package for this purpose). Opening just plain "html" from your harddrive won't work because of security settings of chrome

fenyvesi, 2021-12-13:
Seems to be working. I will try the local web server version, too.

gabonator, 2021-12-13:
When I start it locally with ```http-server -p 8876 .```
and open `localhost:8876` it will show:
```
Failed to load resource: the server responded with a status of 404 (Not Found)
DevTools failed to load source map: Could not load content for wasm://wasm/rtl433.wasm.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
```

The first message is related to missing favicon.ico and the second one is missing wasm.map which is just debugging file for web assembly package. Both are harmless. There is some issue in your case, even the webusb subsystem is not available. Try using different server

gumalexgum, 2022-03-19:
Hello! I really liked your product, but I can't seem to get it set up correctly. Help me to understand. I am not very strong in STM 32 and have not worked with them. It would be ideal to make a video tutorial on how to flash LA 104 in Windows. I'm using Windows 7 x 64 operating system and VirtualBox with Unix 64. At first I was able to write the sistem .hex file, but then the connection was lost while transferring files and the screen shows: "Uknown section name !" "Assertion failed in .. /sourse/main/Execue.cpp [355] : sectionType > 0, Application halted. Virtual disk 2 Mb starts with first button pressed. But file transfer often fails.

[image]

[image]

gabonator, 2022-03-19:
@gumalexgum this thread is related only to webusb connection issues, please start a new issue. It seems that the `shell.elf` file is corrupted. Start the device with 4th button pressed (or 3rd, I can't remember exactly) to enter safe mode. Delete everything from the usb drive and copy the `shell.elf` again.