gabonator/LA104 #78: Wildcat on the DS213 [possibly feature request?]

b3-4r, 2021-09-02:
A foreword: I'm fully aware that this is a stupid question and that this is not the right place to post about it. The LA104 OS as well as the available applications are amazing and I'm impressed by the ingenuity and effort that went into making them possible.

Please forgive the following noobish, desperate question.

I've recently upgraded from the DS203 to DS213 to make use of the higher bandwidth and sampling frequency. I've been using the Wildcat firmware on my DS203, making extensive use of the FFT functionality. Unfortunately the development on the Wildcat FW seems to have gone stale and there is no official support for the DS213. As a 100% not-a-programmer there is nothing I can do to get it to run on the new scope. Out of fear of bricking the new scope I haven't attempted flashing Wildcat nor running it using the LA104 OS.

Would it be possible to port Wildcat to the DS213 platform or better yet to run it as-is? Are there major issues or roadblocks that would prevent this completely? Can I possibly help in any way?

Thank you! 💙

darkonerakot, 2021-09-12:
I think wildcat is not what we need right now. The current OS is pretty good. Bug fix in fpga  that's the issue.

gabonator, 2021-09-12:
There is a bug in the FPGA code in DS213 and the manufacturer did not release any fix nor the source code... I asked them many times if if they are working on it, but I have bad feeling that they gave up. Probably focusing on other products. For me it is very important to have full control over the device, without fully working FPGA chip I do not have any motivation to develop more advanced oscilloscope app. I already tried porting wildcat fw (or another one, cant remember exactly which), but this fw packs too many features into single app and I could not fit it into the OS. The idea of having OS on those little boxes is to have many simple apps instead of one feature-rich app. Currently I am working on ["gabuino"](https://github.com/gabonator/LA104/tree/master/system/apps_featured/117_gabuino) which will be arduino-like development environment for LA104/DS203/DS213. This could be interesting project for other DS213 owners - to make own data logging or visualisation apps without setting up building environment...

gabonator, 2021-09-22:
Bad news guys, I have just got reply from MiniWare. They say they have some NDA with the FPGA supplier which prevents them from sharing any information regarding the software they use for compiling the FPGA code nor the code snippets which could help us develop our own FPGA code. The communication with the supplier is as they say "not good", which indicates that our hope of releasing fixed version of FPGA code is lost. That would be great surprise after waiting for 2 years anyway... 
They are planning to release new version of DS213 with different FPGA chip where more support will be provided for us.

This is really strange and for me there is just one reason to keep everything in secret - AG1KLP could be fake clone of some well known FPGA chip. I remember reading in the documentation that "ordinary FPGA synthesis tools" could be used for programming these chips. So if we do have some FPGA engineer here willing to reverse engineer the FPGA code, please let me know.

But from official perspective this is definitely the dead end for DS213.

darkonerakot, 2021-09-22:
I think current

darkonerakot, 2021-09-22:
I think the current fpga has a hardware malfunction which cannot be fixed. If it reveals thousands of customers gonna ask refunds. But there is a lot of people who don't need all this functionality of the DS. I think this is fraud. But for some people it is good for its price. @gabonator I started learning programming. Your work inspired me.

gabonator, 2021-09-22:
The FPGA works nicely, only in specific cases it does not trigger properly. I believe it is just some minor bug in the code which could be fixed easily if we had the source code. But anyway I do not like the design of DS213 - it has two FPGAs for sampling the ADC separately for both channels. This approach does not allow writing decoders which work with both inputs at once (e.g. USB decoder).
And yes, it is huge issue if they will admit that this device is just faulty and cannot be fixed. So I am sharing this message from them quietly here in this discussion hoping it wont hurt their sales. Most of their customers are not very skilled hw engineers and for simple analysis and fun the device is perfectly sufficient.

@darkonerakot don't forget to share with us your early programming experiments :)

darkonerakot, 2021-10-03:
> The FPGA works nicely, only in specific cases it does not trigger properly. I believe it is just some minor bug in the code which could be fixed easily if we had the source code. But anyway I do not [...]

I am programming with Keil and as I know you use gcc, I was wondering why did you choose gcc, I use only cmsis driver to understand basics principles of interfaces, sorry for off top, I just started, I wish I could share my progress but I don't know how to do it.

b3-4r, 2021-10-03:
Thank you all for shedding some light on this issue. To say this is a bummer is an understatement and situations like this only affirm my belief in truly free(-as-in-speech) software. Guess I'll sell the DS213 and go back to the DS203 even though it's still a considerable downgrade in bandwidth.

darkonerakot, 2021-10-03:
> Thank you all for shedding some light on this issue. To say this is a bummer is an understatement and situations like this only affirm my belief in truly free(-as-in-speech) software. Guess I'll sell the DS213 and go back to the DS203 even though it's still a considerable downgrade in bandwidth.

I didn't find ds203 on market.

b3-4r, 2021-10-05:
> I didn't find ds203 on market.

Oh I still have mine. I don't usually let go of tools when I upgrade unless they're broken or too expensive, the value of having a backup to me is higher than getting some money back for the used stuff.