gabonator/LA104 #82: WebUSB unable to see device on ChromeOS ChromeOS

qwandor, 2021-12-04:
I recently got an LA104 and flashed it with the [latest release](https://github.com/gabonator/LA104/releases/tag/2021.03.24) of this firmware, hoping to use the WebUSB RF analyser functionality. When I connect it to my Android phone and open https://rawgit.valky.eu/gabonator/LA104/master/system/apps/79_cc1101osc/web/index.html in Chrome, it manages to connect (but is not very usable on a phone screen). However, in Chrome on either my Chromebook or Mac laptop, when I try to connect it says there are no compatible devices available. This appears to be due to a permissions issue, as Chrome is not allowed to open the raw USB device.

Is there an easy way around this?

I did notice that using the WebSerial example at https://unjavascripter-web-serial-example.glitch.me/ I was able to successfully connect, so maybe it would make sense to use WebSerial rather than WebUSB?

qwandor, 2021-12-04:
Or maybe it would be possible to support both WebSerial and WebUSB, using the [WebUSB polyfill](https://github.com/google/web-serial-polyfill)? I don't really know Javascript unfortunately, so I'm probably not the best person to do this.

gabonator, 2021-12-04:
It should work on Mac, I am using it regularly (not directly the rf analyser, but other projects that use webusb). After starting the app try reconnecting the usb cable (both ends) - the system needs to re enumerate the device. This could be done by software resetting the USB peripheral, but I did not know how to do it on STM32F103. Regarding the webserial - I need to do some research on this, I haven't tried it. Webusb seems more suitable for such applications, because it will directly open the right web application. 

Please try reconnecting the usb cable more times... and let me know if it helped (on osx)

qwandor, 2021-12-04:
Huh, interesting. I started the app before plugging it into my Mac and it definitely didn't work, but when I unplugged it and plugged it back again it did work. So I guess it's just ChromeOS that's not working (and Linux, unless I add custom udev rules).

wormyrocks, 2021-12-14:
Also not enumerating on macOS (device stays on "Connect the device to computer" screen).
All firmware built from latest commit (980270baf...)

lsusb output:
`Bus 002 Device 001: ID 1209:db42 1209 DS203 Oscilloscope  Serial: 257005434E55393136FFD50`
(I should add that I have a DS213, not a DS203.)

gabonator, 2021-12-14:
@wormyrocks, what application are you using? What is shown in chrome's console? Please try 117_gabuino, the most recent version works well on my M1 (bigsur 11.6). If it won't work I will try updating my mac