gabonator/LA104 #83: LA104 How to format eeprom/flash if cant open LA104 disk at any O/S

r3df0xx, 2021-12-07:
LA104 memory 8,4mb overfilled, 0 bytes free. After power on system cant load properly and stucks at rendering only 1 icon and no actions possible. In DFU mode loads OK.  How can I format or erase memory to re-flash the system and then image with apps i'v built ?

gabonator, 2021-12-07:
Copying files one by one into the eeprom is slow and wears out the eeprom chip possibly leading to its damage.  It is much better to transfer whole filesystem image to the device in one single step. For this purpose there are following scripts:
https://github.com/gabonator/LA104/blob/master/system/scripts/imagefile_make_la104.sh
https://github.com/gabonator/LA104/blob/master/system/scripts/imagefile_copy_la104.sh

If you have your device attached to computer during it's startup, the computer will scan it and write some garbage there. During first run the gui shell application generates "thumbnails" of icons and when both computer and the device are writing to the eeprom, the filesystem gets corrupted (FAT table to be more precise). After copying the filesystem (either file by file, or by copying whole filesystem image) disconnect the device from computer and restart it. Go through all menus and highlight all icons to generate all of the thumbnails.

To sum it up:
- learn how to transfer whole filesystem image (it is faster and more safe than formatting and copying file by file)
- before first run disconnect the device from computer
- go through all icons in the gui

r3df0xx, 2021-12-08:
I solved this issue: erased eeprom via SAFE MODE (hold ESC button on while powering ON). Anyway appriciate your fast reply. Could you provide me instructions how to properly install apps to catch supported radio keyfobs save them and replay (transmit) ?

closed 2021-12-08 by r3df0xx