gabonator/LA104 #84: LA104 release 24 Mar 2021 the Application DS1820 temperatur grapher works not !!

skink32, 2021-12-09:
@gabonator,
I have downloaded the latest version of the LA104.
All tools work except for the DS1820 temperature grapher tool.
I copied it then from version 13 Nov 2020 the 3 files (dstemper.bmp,dstember.elf,dstember)of the apps to version 24 Mar 2021 then it works.
Maybe you can look over there again!?

skink32, 2021-12-09:
Also in the last version 23 Dec 2020 of DS213 are some bugs.
Maybe you can also look over there again!?
Would be great.
Thank you for the super great tools and features that you have integrated into the two devices!
Greeting skink32

gabonator, 2021-12-09:
How did you connect the temperature sensor to the LA104? What does the application show on the screen when you run it?

skink32, 2021-12-24:
@gabonator,
I wish you first of all Merry Christmas !!!

If you have time again it would be very nice if you could compile the files again. So that I could then download them again only as "system_la104.hex" and "la104apps.zip" !
Since I have no idea about programming.

Thanks for your efforts in advance.

Merry Christmas and stay healthy 

skink32

gabonator, 2021-12-24:
Merry Christmas to you as well!

I am currently in covid isolation and have only DS213 with me, so I cannot release new version officially since I cannot cannot check it, but here is a fresh build for you:

[firmware211224.zip](https://github.com/gabonator/LA104/files/7774317/firmware211224.zip)
[la104apps211224.zip](https://github.com/gabonator/LA104/files/7774318/la104apps211224.zip)

Please check the DS1820 app and if it will work consider closing this issue

skink32, 2021-12-25:
@gabonator,
that doesn't sound good!!!
Stay healthy and get back on your feet quickly.

I have unzipped both their files and put them on the LA 104.
1. the .hex file in dial-up mode does not work after installation.
"A CRC error message comes up and no menu appears".

2. the apps file works after installing to the USB drive, but also not all programs run 100% the DS 1820 app works but now.

I hope it didn't hit you too hard and you will be fine soon.

gabonator, 2021-12-25:
I am fine, it was nothing serious. CRC check is new feature of OS to check its integrity. It only shows warning message when something is wrong. But again I cannot verify whether is not a bug in the code that causes false indication (on DS213 the same code works well). It is normal behaviour when the menu does not appear after clean flashing. You need to copy all files from zip archive - menu is an application on the internal disk drive. Which apps do not work?

skink32, 2021-12-26:
So I have flashed the software on the LA 104.
After I have installed the firmware .hex in dial-up mode, the CRC message appears. And I can no longer call the normal memory in the LA 104. That is, after flashing, I turn off the LA 104 and on again but it no longer appears the 8GB large memory area in which the apps are stored.

Have then again taken the .hex of the old version and then installed the new apps. Then went already times the DS 1820 software. The remaining apps I will test later again and write to you or make pictures where you see what does not work. (Of course I always installed all files from the Apps folder, of course).

P.S. Can I install this software on the DS213, but there I can not use many programs because of missing connections!

gabonator, 2021-12-26:
How do you copy the firmware to the device in DFU mode? If the flashing was successful, a file with the same name but different suffix should appear on the DFU drive. I cant remember what was the suffix, but if the suffix is ".ERR", it means that flashing went wrong. Try copying the .hex file two times: 
1. copy the file to dfu, 
2. wait until the disk reconnects,
3. copy the same file again,
4. check the file name on the drive after it automatically reconnects
Or try flashing on another computer.

skink32, 2021-12-26:
So I copy the .hex file in DFU mode, but after copying should actually appear the same file with the extension .rdy (for Ready) that does it but neither with the old nor with the new version!

After you have restarted then you come normally in the memory mode so that you can access the 8GB internal hard drive. This works but with your .hex file that you have provided me then no longer after I have transferred you in DFU mode!
So after I have transferred the .hex no longer the normal memory 8GB displayed no matter how often I switch on or off!

It can not be the computer, because it works with all other and the files I changed (Apps file) also without problems.

gabonator, 2021-12-26:
Press first button when it shows the CRC error message. It will just continue running and it should turn on the usb mass storage mode.

Ziedins15, 2022-08-04:
> Press first button when it shows the CRC error message. It will just continue running and it should turn on the usb mass storage mode.

Exactly!

 How i flash LA104:
1. Connect LA104 to PC
2. Switch on LA104
3. Clean content of 8Gb memory in mass storage mode. (After that don't try operate LA104 menu)
4. Copy fresh APP content on 8Gb memory
5. 5.witch off LA104.
6. Boot in DFU mode (keep pressed first button and switch on LA104)
7. In PC appears DFU drive, copy on it new system .hex file (Wait till drive reconnect)
8. Switch off LA104
9. Switch on LA104 and wait some seconds, if CRC appears simply press first button.
10. You are in new system.

For me, after system .hex re flashing, DFU drive don't will reconnect. It simply disconnect. For this reason i don't now status of flashing. I think that is my PC USB host problem. After flashing i simply switch off and then on device.

P.s. - CRC appears for me too

Can somebody provide me with 1-Wire sensor connection diagram to LA104. Thanks

justsux, 2022-12-14:
I can confirm that this is resolved with the most recent build.
Thank you @gabonator !

fix: https://github.com/gabonator/LA104/releases/tag/2022.12.12