gabonator/LA104 #88: Problem with LA 104 firmware via USB

gumalexgum, 2022-03-19:
[](https://translate.google.com/website?sl=auto&tl=ru&hl=ru&anno=2&u=https://user-images.githubusercontent.com/101945675/159135237-53a87d3e-10fe-4271-a281-e4d23a0ff767.png)[](https://translate.google.com/website?sl=auto&tl=ru&hl=ru&anno=2&u=https://user-images.githubusercontent.com/101945675/159135137-d4483dd1-3e3b-4b1b-b1f5-7db27d174263.png)Hello! I really liked your product, but I can't seem to get it set up correctly. Help me to understand. I am not very strong in STM 32 and have not worked with them. It would be ideal to make a video tutorial on how to flash LA 104 in Windows. I'm using Windows 7 x 64 operating system and VirtualBox with Unix 64. At first I was able to write the sistem .hex file, but then the connection was lost while transferring files and the screen shows: "Uknown section name !" "Assertion failed in .. /sourse/main/Execue.cpp [355] : sectionType > 0, Application halted. Virtual disk 2 Mb starts with first button pressed. But file transfer often fails.

[image]

[image]

gumalexgum, 2022-03-19:
> @gumalexgum this thread is related only to webusb connection issues, please start a new issue. It seems that the shell.elf file is corrupted. Start the device with 4th button pressed (or 3rd, I can't remember exactly) to enter safe mode. Delete everything from the usb drive and copy the shell.elf again.

gumalexgum, 2022-03-19:
Safe mode.  Windows: USB device not recognized!

gumalexgum, 2022-03-19:
> @gumalexgum It seems that the shell.elf file is corrupted. Start the device with 4th button pressed (or 3rd, I can't remember exactly) to enter safe mode. Delete everything from the usb drive and copy the shell.elf again.

Sorry! Does not work! The USB device is not recognized by Windows.

gabonator, 2022-03-19:
Show me screenshot of device manager. If the drive is just not accessible, try formatting it

gumalexgum, 2022-03-19:
> Show me screenshot of device manager. If the drive is just not accessible, try formatting it

[image]

gabonator, 2022-03-19:
That is weird, can you try attaching it to another computer?

gumalexgum, 2022-03-19:
Yes, I tried connecting to a Windows 7 x 32 laptop. Same thing. In my opinion, there was a failure during the transfer of files and an incorrect file name. Because of this, the USB is not connected.

gumalexgum, 2022-03-19:
When copying system_la104.hex, system_la104.err appears first, which I will rename to system_la104.hex, after which system_la104.rdy appears. I don't know if this file is correct or not?

gabonator, 2022-03-20:
No, do not rename them. Just copy for the first time, wait util the device gets disconnected, then copy the file again.

gumalexgum, 2022-03-22:
Can I ask you to write down the points that I need to do and in what order?

one) ...
2) ...
3) ...
...

gumalexgum, 2022-03-22:
[image]

gumalexgum, 2022-03-23:
What USB device should LA104 detect when connected to a computer? And why does Windows 7 not detect the device ID?

mgazza, 2023-12-26:
people seeing this issue cd to the LA104 drive and run `. -type f -name '.*' -delete`