gabonator/LA104 #97: LA104 Open sesame implementation

dborras33177, 2022-11-25:
Hello, have you ever thought about implementing open sesame (by Sami Kamkar) on this platform LA104?

SpudGunMan, 2022-11-25:
Just sitting on the side and my first thought is, how many hand held wireless devices need to be hacked to steal cars and open garage doors.

gabonator, 2022-12-07:
Look at this part of rf toolkit demo video: https://youtu.be/Gwyi00NKBNg?t=184
I am pressing the garage door remote several times and you can see that the waveform payload is mostly static code and only single byte incrementing counter is changed every button press. So everything what you need is to just add few lines of code into RF toolkit app which would generate all 256 codes and transmit them in sequence - no need to have separate app for that. 
Porting the open sesame program should not take more than one day, but it is currently not on my LA104 project road map. If anyone will port that program, I will be happy to include it in the la104 release package.