gabonator/LA104 #103: LA104 SPI Problem: MSB and LSB

magicroomy, 2023-01-12:
I noticed by viewing the waveform of SPI communications and erifying with my Logic Analyser that the configuration of MSB/LSB in the SPI decoder are reversed.
MSB should be LSB and LSB should be MSB.
I am not sure if this issue an be solved here, but I tried to place an error report somewhere at miniware. My Chinese is not good enough :-(

gabonator, 2023-01-13:
Please send screen shot

magicroomy, 2023-01-13:
Hi, attached you find a picture of the settings and the waveform.

The sequence is generated by an ESP32. 

With this code snipplet

 

SPI.beginTransaction( SPISettings(200000, SPI_MSBFIRST, SPI_MODE0)) ;

digitalWrite(5, LOW) ; // This is CS!

SPI.write(0xBB) ;

SPI.write(0x11) ;

digitalWrite(5, HIGH) ;

 

As you can see: SPI is configured to have 1’s on high level and data on the positive clock.

So the Bits are 

1 0 1 1 1 0 1 1

MSB = BB

LSB would be DD

The problem is as stated: Setting is LSB, but the analysed data is showing BB.

 

In the meantime I wrote to a completely generic Miniware Email and indeed got an answer. They will have an engineer check it and if I’m right provide a patch.

 

Cheers

Volker

gabonator, 2023-01-13:
Where is the picture?

magicroomy, 2023-01-13:
Here they are.
The SPI output was created by a ESP32 set to SPI Mode0 and MSB

[image]
[image]

gabonator, 2023-01-13:
When you switch to MSB, will it show 0xDD?

magicroomy, 2023-01-13:
Yes it does

gabonator, 2023-01-13:
Bit order in decoder:
https://github.com/gabonator/LA104/blob/master/system/apps/test17_official/source/Analyze.c#L29
MSB/LSB constants:
https://github.com/gabonator/LA104/blob/master/system/apps/test17_official/source/Analyze.h#L12
Menu:
https://github.com/gabonator/LA104/blob/master/system/apps/test17_official/source/Menu.c#L994

`gInputSubParam[IN_SPI][3]` holds the information what kind of signal we are going to decode:
```
memcpy(&param_str[LEN * 3], (gInputSubParam[IN_SPI][3] ? " MSB " : " LSB "), SIP_PARAM_STR_LEN);
```
where 0 means LSB and 1 MSB
```
#define  LSB           0   // 
#define  MSB           1   // 
```

`SPI_AnalyzeFrame` does the decoding part:
```
                if(CLK_Phase == 1)                         // 
                {
                    if(SCK_Now == CLK_Polarity)
                    {
                        AnalyzeBuf[j * 3 + 0] |= BIT_Order ? ((Status & 0x01) << k) : ((Status & 0x01) << (7 - k));
                        AnalyzeBuf[j * 3 + 1] |= BIT_Order ? (((Status >> 1) & 0x01) << k) : (((Status >> 1) & 0x01) << (7 - k));
                        if(k < 7) k++;
                    }
                }
                else                                       // 
                {
                    if(SCK_Now != CLK_Polarity)
                    {
                        AnalyzeBuf[j * 3 + 0] |= BIT_Order ? ((Status & 0x01) << k) : ((Status & 0x01) << (7 - k));
                        AnalyzeBuf[j * 3 + 1] |= BIT_Order ? (((Status >> 1) & 0x01) << k) : (((Status >> 1) & 0x01) << (7 - k));
                        if(k < 7) k++;
                    }
                }
```

`k` is incrementing from 0 to 7, it holds the bit index. This expression `BIT_Order ? ((Status & 0x01) << k) : ((Status & 0x01) << (7 - k))` should evaluate the bit weight, probably all of these 8 partial values are later summed together to get transmitted/received value. When decoding LSB frame, the first bit should give value 1 and 128 for MSB respectively. For LSB the `Bit_Order` is zero (false), so the second term is evaluated `((Status & 0x01) << (7 - k))` which is wrong.
I recommend changing `BIT_Order ? .....` into `(BIT_Order == LSB) ? ...` which will invert the condition to fix this problem.

gabonator, 2023-01-13:
Try this:
[17analys.zip](https://github.com/gabonator/LA104/files/10414738/17analys.zip)

magicroomy, 2023-01-13:
I renamed the original analyser.elf to enalyser.elf___ and took the content of your 17analys.zip and put it into the device as analyser.elf
Restart of the device. analyer started.. I am afraid. Exactly the same display.. BB on setting LSB ;-(

gabonator, 2023-01-13:
`!BIT_Order`, `BIT_Order == 0`, `BIT_Order != 1`, `BIT_Order == LSB`, `BIT_Order ~= MSB`, all are equal. But ternary operator is a bit tricky, because you never know, if the negation operator `!a ? b : c` is applied on the first argument, or on the result, so it is always safer to put the condition in parenthesis `(!a) ? b : c`

gabonator, 2023-01-13:
I didn't notice that there are 8 places with this code, not four. Try this
[17analys.zip](https://github.com/gabonator/LA104/files/10415251/17analys.zip)

magicroomy, 2023-01-13:
yeah you are abolutely right with the parenthesis. 
AND ! Tadaaa. I seems to work now. I owe you something

magicroomy, 2023-01-13:
Funny enought after YEARS of the LA104 being on the market I am the first to notice??? Hmmm :-)

gabonator, 2023-01-13:
I would really like to know how many units did they sell and how many are actually used... but thank you for reporting this issue

magicroomy, 2023-01-13:
Glad that you could help me :-)

gabonator, 2023-01-13:
You are welcome :) But why are you at friday night fighting with logic analyser and not somewhere out with friends and drinking beer? :)

magicroomy, 2023-01-13:
Poor me.. I am sick at home while they have one after the other :-(  (pity is welcome)

gabonator, 2023-01-15:
Fixed here https://github.com/gabonator/LA104/commit/a8904d5b59a4e18aa657cb5ea7204a57d635f92b

closed 2023-01-15 by gabonator