gabonator/LA104 #106: DS213 Web Oscilloscope application gives no data

Martyx00, 2023-01-19:
Hi,
I have successfully installed the latest build for the DS213 (https://github.com/gabonator/LA104/releases/tag/2020.12.23). However, I cant make the Web Oscilloscope to work. When I select the app upon the boot and connect the device to computer, I get the pop-up which opens the application, but than nothing happens, no data appear.
The device is registered under `/dev/ttyACM1`. I am using Ubuntu 22.10, are there any tricks that I have to do to make it work?

```
[28737.107080] usb 1-2: new full-speed USB device number 23 using xhci_hcd
[28737.287040] usb 1-2: New USB device found, idVendor=1209, idProduct=db42, bcdDevice= 2.20
[28737.287052] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28737.287056] usb 1-2: Product: DS203 Oscilloscope
[28737.287059] usb 1-2: Manufacturer: Gabriel
[28737.287062] usb 1-2: SerialNumber: 037302574241303939FFD70
[28737.341270] cdc_acm 1-2:1.0: ttyACM1: USB ACM device
```

Martyx00, 2023-01-19:
Sending any character via serial causes Hard Fault and register dump on the screen.
[image]

gabonator, 2023-01-19:
Try this
[webosc213.zip](https://github.com/gabonator/LA104/files/10461548/webosc213.zip)

Martyx00, 2023-01-20:
The issue is the same (except that the PC during the crash is now at `0x80644D0`).

gabonator, 2023-02-01:
I have just tried it and everything works well - at least on OSX. Make sure you have following page opened: `https://rawgit.valky.eu/gabonator/LA104/master/system/apps_featured/69_webusbosc/web/v3/index.html`

But I confirm that it crashes when I send a single character through serial link with 
`screen /dev/tty.usbmodem101 9600`
as well as with
`minicom -D /dev/tty.usbmodem101`

This is a bit strange and I will investigate this issue following days

Martyx00, 2023-02-01:
Just made it work, I had to add permissions for USB control to my current user:
```
echo 'SUBSYSTEM=="usb", MODE="0660", GROUP="plugdev"' > /etc/udev/rules.d/00-usb-permissions.rules
udevadm control --reload-rules
```
Than just add the user into the `plugdev` group and it should work.