gabonator/LA104 #107: LA104 WebUSB connection problem 2

fenyvesi, 2023-01-22:
Latest application (2022.12.22), Win10
1. Starting WebUsb RF analyzer
2. Connecting USB. (Device manager: LA104 device, COM9.)
3. Starting Chrome, nothing happens.
4. Manually open https://rawgit.valky.eu/gabonator/LA104/master/system/apps/79_cc1101osc/web/index.html
5. Push Connect button, select LA104 Analyzer from the window, connect
6. Nothing happens.
7. Chrome console: 
index.html:1 Uncaught (in promise) DOMException: Failed to execute 'open' on 'USBDevice': Access denied.
Promise.then (async)
connect @ comm.js:29
(anonymous) @ comm.js:66
Promise.then (async)
(anonymous) @ comm.js:64

May be I made something wrong. Advice?
Thank you.

gabonator, 2023-01-22:
https://zadig.akeo.ie/ use this to generate webusb compatible driver for la104

fenyvesi, 2023-01-22:
WinUSB driver installed, works,
Thank you.

closed 2023-01-22 by fenyvesi