gabonator/LA104 #108: Strange pattern shown on "desktop" when exiting Logic analyzer app

szmolnar, 2023-02-05:
Hi!

I recently uploaded the recent release 10.
Everything went ok, no issues at all but I'm getting problems when exiting the Logic analyzer app, the background changes.
Can't restore the original background in any way but switching off the device than back on.

Also tried to re-format the device and install everything again but got the same results.

Background Before starting the app:
[image]

Background After exiting the app:
[image]

Device info:
[image]

i also wish to use PulseView but I can't find the USB application which emulates sigrok SUMP protocol, it is missing from the recent build.

Thanks!

vochong, 2024-02-09:
Hi szmolnar,

How do you exit the Logic Analyzer app?   Pressing Menu button does not work like the other apps.  I just have to power off/on.

vochong, 2024-02-10:
I have the same issue.