gabonator/LA104 #115: Diff. between Flipper Zero and LA104

loewal, 2023-07-08:
Hello Gabriel,

Why do I get those strange values from my LA104?
See picture.

Regards!
[image]

gabonator, 2023-07-08:
Cant see the picture

loewal, 2023-07-14:
[image]

loewal, 2023-07-14:
Like this. Sometimes the shown temperature and humidity values are even stranger, like -214 or so...

loewal, 2023-07-14:
Tried to recalculate into Kelvin, Fahrenheit etc, but no ressemblance

loewal, 2023-07-14:
[image]

loewal, 2023-07-14:
-200.0 C...?
Flipper is correct

gabonator, 2023-07-14:
Clearly the application has chosen wrong decoder, the bitstream is received correctly but wrongly interpreted. I have only implemented a few of decoders into the application since I did not have other devices at home to test. Please try to capture the waveform:
https://youtu.be/Gwyi00NKBNg?t=248

If you open "developer console" in chrome, it should print the timing in microseconds (e.g. [200, 400, 400, 200, 200, 400...])
This project is open sourced and anyone with a little C++ skills can any bug they found. Unfortunately I do not have capacity to fix this issue, but at least we can identify the right decoder for your temperature sensor

loewal, 2023-07-14:
I get this:
analyse: [520,8900,480,4060,500,2000,500,4040,500,4040,480,4060,500,2020,480,4040,500,4020,520,2040,480,2040,480,2040,480,2060,500,2020,480,2040,480,2100,500,2020,480,4060,480,4020,500,4060,480,4020,500,2060,480,2020,520,4020,520,2020,480,2060,480,4020,520,2000,500,2040,500,4020,480,4040,500,4020,520,4100,500,4040,480,4040,500,2020,520,2020,480,4040,480,8900,500,2120,480,8900,500,4040,520,2020,500,4040,480,4020,500,4100,500,2040,480,4060,480,4040,500,2020,500,2060,480,2040,500,2040,500,2020,480,2060,480,2020,500,2040,500,4040,460,4060,500,4080,520,4020,500,2020,480,2040,500,4100,500,2020,500,2040,480,4040,520,2040,480,2060,480,4020,500,4020,520,4040,500,4020,480,4040,500,4040,480,2040,480,2040,480,4060,500,8920,500,2120,500,8880,520,4040,500,2020,480,4040,520,4020,480,4060,480,2060,460,4060,480,4060,480,2060,480,2040,500,2040,480,2040,500,2020,500,2000,520,2080,500,2060,460,4060,500,4020,480,4040,520,4020,520,2040,500,2040,480,4060,480,2040,500,2020,500,4040,500,2020,500,2040,500,4020,480,4040,500,4040,520,4080,480,4060,480,4060,500,2000,500,2040,500,4040,460,8900,480,2140,520,8880,480,4100,500,2040,500,4040,480,4060,480,4100,480,2040,500,4060,460,4040,500,2040,480,2040,500,2020,480,2040,500,2040,460,2040,520,2020,520,2000,500,4020,520,4020,520,4020,480,4060,500,2020,500,2000,520,4120,480,2040,500,2000,520,4020,500,2020,500,2040,480,4020,520,4040,480,4040,520,4020,480,4060,480,4040,520,2040,460,2040,500,4020,500,8920,520,2140,460,8920,480,4060,480,2040,500,4040,500,4000,500,4060,500,2040,480,4040,500,4040,480,2060,500,2040,460,2060,500,2040,480,2060,500,2020,500,2040,480,2040,480,4080,480,4020,500,4020,500,4040,480,2060,500,2040,500,4020,480,2040,500,2040,500,4020,500,2040,480,2040,480,4060,480,4060,460,4060,480,4040,500,4040,500,4060,500,2020,500,2000,500,4060,500,8860,520,2120,480,8880,500,4040,500,2040,500,4020,500,4040,500,4120,480,2040,480,4020,520,4040,500,2040,480,2020,520,2000,480,2040,500,2040,520,2040,500,2020,500,2000,500,4040,500,4020,500,4020,520,4040,500,2020,480,2040,500,4060,460,2040,520,2000,480,4080,480,2060,480,2020,500,4060,480,4060,480,4020,520,4020,500,4060,480,4060,480,2040,480,2040,480,4040,500,8960,480]

But none of the received signals gives a known name like GT-WT02 37b

And helas, I am a former programmer...of Cobol, Pl1 and a very little bit of assembler ans also ages ago.
It was nevertheless very interesting to struggle with the LA104.

Thank you for such a clever bunch of applications!

loewal, 2023-07-14:
But its strange I guess, that  the RF_analyser_synthesizer gives some kind of result....shouldn't it do nothing...?

loewal, 2023-07-14:
Thank you Gabriel for elaborating on this interesting subject.
Patreon nothing for you?

gabonator, 2023-07-15:
Patreon sound like begging to me. The LA104 community is not that big to make this profitable. I consider this project more sort of presentation of my skills for future employer - as tons of other projects I publish on github. Everything I did here was done for fun and to learn how to make simple operating system (or application loader with rich C++ API) and to see how much can I get from STM32F103. And by this project I also fulfilled my dream from highschool - to make an electronic device which can talk to any component/sensor/TV over infrared. Simple and cheap universal tool to replace expensive lab equipment.

loewal, 2023-07-17:
It is indeed impressive what you developed.
One last question:
It happened already several time, that my LA104 could not  load shell.elf and hung.
Quite a process to get verything back to normal, because sometimes the copying of programsand folders stops halfway.
I discovered that safemode (key 3 ESC) is more steady.
What can I do to prevent this situation?
Make some files readonly perhaps....or is the LA104 quite vulnerable to errors?
By the way: I always do a format.

gabonator, 2023-07-17:
I know about that issue. The problem is when the device is connected to a PC and both - the device itself and PC - write some data to flash disk. The PC initially scans the drive and keeps the FAT table cached in memory. When you are browsing the icons in GUI for the first time, the device saves a "screenshot" (file suffixes tm0, tm1, tm2, tm3) of each icon you highlight to speed up rendering. But later the computer decides to write something unaware that the FAT table was changed... and this breaks the filesystem. I have prepared a solution for this, instead of placing BMP icons on the drive, I generate the "screenshots", so the device itself should not write anything to the disk. But I did not release this fix yet. My suggestion is, that after copying all the files from PC to LA104, disconnect the device, turn off and on for the GUI to start up and go through all icons in every folder (just highlight every icon, no need to run any app). This should help

If you are running linux or OSX, there are scripts `scripts/imagefile_make_la104.sh` and `scripts/imagefile_copy_la104.sh` which will reinstall whole filesystem

loewal, 2023-07-17:
Thanks again Gabriel.
One addition: connecting the LA104 to a Mac computer generates a load of ._ files. 104!
I ran a nice app, Blueharvest (free) that scans for those files, also Spotlight files, and deletes them and subsequently unmounts the "drive".

loewal, 2023-07-20:
Hi again,
Could you show me a picture of how you soldered the connector for the cc1101?
Thanks

loewal, 2023-07-24:
Hi Gabriel,
Is it somehow possible to help me on the right way to add some RTL-433 encodes?

gabonator, 2023-08-05:
connection guide: https://github.com/gabonator/LA104/tree/master/system/apps_featured/80_rftool
In a few weeks I will be working on a youtube video showing how to capture unknown signal and how to add decoder into rftool, so you could use it as a inspiration

McGr3g0r, 2023-08-09:
Hello, for rtl_433 protocols I guess I could contribute as I already did with HCS200 and EV1527 implementation.
As Garbriel mentioned, there will be issue running all the protocols at once - maybe we could build many versions of rftool with split / disabled protocols an choose a proper one with app manager (I have it this way on my local build already) ?