gabonator/LA104 #118: Major refactoring needed

salco, 2023-08-08:
I've cloned this repo last week and I had to change many apps (all had same issue: type declaration inside a header).
 it seems that they are all a copy paste of certain header and source.

My question is: @gabonator with the use of Cmake and the library can I try to refactor the project?
It could help adding new apps, upgrading gcc, automate some process, even add unit test.

gabonator, 2023-08-09:
Feel free to do it. I am using cmake only for generating XCode project for emulated environment for some applications. I do not think it is suitable for building whole OS. Upgrading gcc doesn't give much sense, there are no concurrency related stuff used here, and also it would break the gabuino cloud based compilation service. Adding more apps is highly welcome.

Yes, I know about that header issue, I have just changed some declaration in library and forgot to fix it in some apps... I am right now busy with other stuff, so I let the community to do further development/fixes

McGr3g0r, 2023-08-09:
Hi,
I had to make many changes such as this to make the whole project compile:

gabonator, 2023-08-09:
Yes, that is exactly it, it should be on three different places I guess, feel free to open PR

McGr3g0r, 2023-08-09:
https://github.com/gabonator/LA104/pull/120

salco, 2023-08-13:
yeah I will test that the change are not lie mine but I've also changed the dockerfile

salco, 2023-08-17:
the dockerfile work fine but for the refactoring I can push it in a separate PR.