gabonator/LA104 #119: LA104 add new protocol into filmware

kopRF, 2023-08-09:
https://github.com/gabonator/LA104/pull/116/commits
I saw that someone is creating a new protocol for the application, understood which changes in the files need to be replaced, but I don't know how to implement it in the firmware to create a ".hex" file. I have Windows, and I would appreciate your help.

kopRF, 2023-08-10:
Hello! Thank you very much for your response 🙂. However, I don't know how to create a new binary file 80rftool.elf using the data you provided me with. Even after you explained the process, I'm still facing difficulties. Perhaps I'm not searching for information correctly on Google or my queries are not accurate. I'm not a programmer, but I'm trying to learn gradually because it's interesting to me. Therefore, its litle dificult for me... :(

gabonator, 2023-08-10:
At first you need to build the OS, then library, then any application you want, just follow this tutorial and let us know which step doesn't work for you:
https://github.com/gabonator/LA104/tree/master/resources/tutorial_building#mac-osx--linux

loewal, 2023-09-29:
Did a complete recompile. Went well!
Then copied 80rftool.elf after renaming to rftool.elf to LA104, overwriting the old one.
Now I get an assertion error.
What am I doing 'wrong'?

gabonator, 2023-09-29:
What is the assertion?