gabonator/LA104 #126: Changes

loewal, 2023-09-27:
Hi Gabriel,
I saw that  you made some changes in the last weeks, e.g.cc1101.
I tried to compile as I always did before, but now I get an error:

/usr/local/Cellar/arm-none-eabi-gcc/10.3-2021.10/gcc/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: cannot find -lbios_la104
collect2: error: ld returned 1 exit status
arm-none-eabi-objdump: 'output.elf': No such file
readelf: Error: 'output.elf': No such file
./build_la104.sh: line 32: ../../../../tools/elfstrip/elfstrip: No such file or directory
/Library/Developer/CommandLineTools/usr/bin/nm: error: output.elf: No such file or directory
/Library/Developer/CommandLineTools/usr/bin/nm: error: output.elf: No such file or directory
/Library/Developer/CommandLineTools/usr/bin/nm: error: output.elf: No such file or directory

Is it asked too much if you can provide a new complete release for the LA104?

Thank you!

gabonator, 2023-09-27:
Which script are you using for building? What does it print when you run `system/scripts/build.sh`? From the log it seems you need to rebuild the library (run `system/os_library/build.sh`) and you are missing elfstrip tool (run `tools/elfstrip/build.sh`) Currently I am working on something that might be interesting for you - visual environment for designing custom decoders/encoders. It should enable anyone to design new decoders without any programming skills. I would like this platform as a database for all encoders/decoders for CC1101 rf tool as well as for new applications. After designing the decoder and checking if it works, it should be able to generate optimized C code with the encoder/decoder. Far from finished, but looks promising, it works with CC1101 webusb app (`79_cc1101osc`). Please check it and give feedback:

https://rawgit.valky.eu/gabonator/LA104/master/system/apps/79_cc1101osc/web/editor/index.html

It is still too early to publish new release, since there is nothing new and I noticed that there are some bugs in the cc1101 web frontend, I would like to fix those bugs and add more decoders at first.

loewal, 2023-09-28:
Thank you for your elaboration!
The frontend looks good, but perhaps too complicated for a layman like me...

I'll wait for your fixes.
Thank you again.