gabonator/LA104 #127: Fresh compile is successful, but error booting

loewal, 2023-10-12:
Hi Gabriel,
I saw that you altered something (rtl_433), so I did a complete build for LA104.
Formatted it and copied all files like I already did several times successfully.
[image]
This is the error I get.

Any clue perhaps?
Regards.

loewal, 2023-10-12:
Worth mentioning. I also tried the latest copy of yourself from dec. 2012.
Same problem.

loewal, 2023-10-12:
Found that MacOS left a .fsevents... and when removed, things worked!
Is it worth the effort to compile again because of rtl_433?

gabonator, 2023-10-12:
No rebuild is necessary. All those changes are related to the signal editor:
https://rawgit.valky.eu/gabonator/LA104/master/system/apps/79_cc1101osc/web/editor/index.html#decode_init;signal_simple2

New node called `Rtl433` was added which can decode signals from more than 240 various devices. This runs inside the web browser, not within the LA104.

loewal, 2023-10-13:
I just wonder how to use this.
- Start the web-app
- stop when finding signal
- select signal
- start web-editor
Is this the way to use it?...because I used above link and saw rtl_433, but not the way I described before....

Will there be a wiki/doc for this?

gabonator, 2023-10-13:
Almost there! Just swap the last two points. When you select the signal, the web app sends it to the child windows. When you open the editor, it will just show some initial testing signal which gets replaced after selecting… documentation is not planned but I would like to do some youtube video how to decode unknown signals. But due to being busy with other stuff this might take a long time

loewal, 2023-10-19:
And working nicely!!
[image]